Ultimaker/Cura #6832: Feature Request: Mouse scroll wheel as layer changer in preview mode

remick-no, 2019-12-20:
Mouse scroll wheel works as zoom which is great.
However in **preview** mode it would be fantastic if after clicking on _layer number field_, rotating wheel (instead of zoom) I could control layer number.
Clicking again outside of layer number field should bring back zoom mode.

Ghostkeeper, 2019-12-20:
The arrow keys do scroll through the layers now, by the way.

But yeah we could consider this as an alternative.

remick-no, 2019-12-20:
I should definitely spend more time reading manuals.
I had no clue about arrow keys function.
Apologies. I guess this "case" can be closed now since there is already solution and no need to spend time reinventing the wheel.

Ghostkeeper, 2019-12-20:
Well, one problem is that there is no manual that explains this sort of thing :stuck_out_tongue:

Ghostkeeper, 2019-12-23:
We've decided to not do this, mostly because you'll probably want to use the scroll wheel for zooming while going through the layers.

closed 2019-12-23 by Ghostkeeper