Ultimaker/Cura #6833: [4.4.0] Cura thinks my keyboard is a printer (Input Club K-Type).

gabfv, 2019-12-20:
**Application version**
4.4.0

**Platform**
Windows 10 with nVidia 1080 Ti.

**Printer**
Prusa i3

**Reproduction steps**
1. Launching Ultimaker Cura.
2. After a short period of time, my keyboard (An Input Club K-Type) enters flashing mode. Pressing ESC on the keyboard does bring it out of the flashing mode, but eventually, it re-enters flashing mode, which is very likely done by Cura.

**Actual results**
My Input Club K-Type keyboard is in flashing mode.

**Expected results**
My Input Club K-Type keyboard **should not be** in flashing mode.

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/3990085/cura.log)

**Additional information**
According to the logs, I think Cura thinks my keyboard is a printer. Is there a way to deactivate this behavior? I don't use Cura with a printer, just for slicing.

gabfv, 2019-12-20:
I've created an [issue](https://github.com/kiibohd/controller/issues/379) with the K-Type firmware as well, since I did not know whether the issue resides in my keyboard or in Cura.

fieldOfView, 2019-12-20:
If you don't use USB Printing, then it is safe to disable the USB Printing plugin via Marketplace -> Installed tab. This will stop Cura from probing any serial ports (thus resetting devices).

gabfv, 2019-12-21:
Thanks, I've disabled the plugin and it fixed the issue. I think I've isolated which log lines occurs when the plugin is enabled: 

2019-12-20 14:25:30,012 - ERROR - [Thread-14] UM.Logger.logException [88]: Traceback (most recent call last):
2019-12-20 14:25:30,014 - ERROR - [Thread-14] UM.Logger.logException [88]:   File "X:\4.4-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 121, in run
2019-12-20 14:25:30,018 - ERROR - [Thread-14] UM.Logger.logException [88]:   File "C:\Program Files\Ultimaker Cura 4.4\plugins\USBPrinting\AutoDetectBaudJob.py", line 84, in run
2019-12-20 14:25:30,021 - ERROR - [Thread-14] UM.Logger.logException [88]:     serial.write(b"M105\n")
2019-12-20 14:25:30,023 - ERROR - [Thread-14] UM.Logger.logException [88]:   File "serial\serialwin32.py", line 315, in write
2019-12-20 14:25:30,025 - ERROR - [Thread-14] UM.Logger.logException [88]: serial.serialutil.SerialException: WriteFile failed (PermissionError(13, 'The device does not recognize the command.', None, 22))

Any chance I can know which USB endpoints are being probed or does it probe them all automatically? This information would help for the other issue I opened. If not, you can close this issue as fixed.

fieldOfView, 2019-12-21:
The Cura USB Printing plugin probes all serial ports automatically.

closed 2019-12-21 by gabfv