Ultimaker/Cura #6834: Prevent Extra Support outside the perimeters

FlailAway, 2019-12-20:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

**Describe the solution you'd like**
No spurious Support towers outside the perimeter of the sliced objetc.

**Describe alternatives you've considered**
PruseSlicer
Ideamaker

**Affected users and/or printers**
Lots of people who use Support I am guessing. 

**Additional context**
Thingiverse item "CaseA" the main body case.
https://www.thingiverse.com/search?q=2129988&dwh=375dfd3deacc3a2

Use stock Cura 4.4.0 .appimage "Draft" Profile with "Support," checked and "Touching Buildplate" then Slice and observe **no** spurious items visible, but way too much Support that should not be needed. 

Set the "Support overhang angle" to 60 to eliminate the unneeded support at the front outside of the case that wastes an hour of print time. Now, observe the stand alone column to the right of the large hole in the right side (looking from the front) of the box. Plus spurious lone columns inside the case near the front.

I have posted this as "Feature request" since my last Bug was downgraded to no-first-layer printed feature, so I may as well start here. {grin}

FlailAway, 2019-12-21:
OK, took some pics.
First two are with Support angle at 50 but it makes that big lump at the front which adds an hour print time.

[image]
[image]

Next 3 show spurious (to me) columns and one outside with the angle set at 65

[image]
[image]
[image]

And, finally using SmartAvionics build and NO Support, but his Bridge settings it looks like he has **nailed** it. I have not printed this one yet but with that short front-to-back bridging I think it will print OK.
[image]

Ghostkeeper, 2019-12-23:
Posting bugs as feature requests is not really going to fool us if those features collide with existing features! I'll try to explain why it's doing these things:

1. In the first screenshot it's creating a bit of support there because of the curved corner. Since the corner is curving away both in the X direction and the Y direction, the added overhang angle is greater than 50% in that corner. I can't reproduce your exact case using Custom FFF Printer defaults, but I get a similar result setting the overhang angle to 61 degrees there.
2. This support pillar is created because of the little ledge starting around 55mm up from the build plate.
3. I couldn't reproduce this problem, but it seems to be similar to an earlier bug report: https://github.com/Ultimaker/Cura/issues/4219
4. Duplicate of this: https://github.com/Ultimaker/Cura/issues/6795

So for 1 and 2 we have two features to remedy this: [Maximum Tower-Supported Diameter](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/support/support_tower_maximum_supported_diameter.md) causes a tower to be generated instead of normal support. This tower prevents the support from toppling over in case it's too thin and tall. This would probably be what you need for the 2nd problem. We also have [Minimum Support Area](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/support/minimum_support_area.md) which can be used to filter small pieces of support out entirely, leaving them unsupported.

Since we've already implemented this feature, I'll close this as being fixed.

closed 2019-12-23 by Ghostkeeper