Ultimaker/Cura #6836: Bluetooth headphones audio interruptions if Cura is running

ylectron, 2019-12-21:
I have an i7 with 8Gb of RAM, hi-end headphones that never caused any issues. 
Only Cura disrupts something in the operating system that makes any audio played by the computer with 30-100 interruptions per minute. Even if I do not do anything in it and it is just waiting in the background makes Youtube crap out.

Ghostkeeper, 2019-12-23:
I'm regularly listening to music on my high-end headphones while developing Cura or just with Cura running in the background, and never heard anything out of the ordinary while it was running, neither on Linux nor on Windows. Depending on how you run Cura in your operating system, Cura should not normally even have administrator access and therefore shouldn't have the power to interrupt your sound driver.

To debug this, we need you to fill in the issue template, please. You can understand that it's useful to know, like... what operating system do you even have that contains this bug? Did you report your bug to Microsoft, Apple or your Linux distribution?

blast-3d, 2020-02-02:
I am having a similar issue with a bluetooth speaker connected to my pc while running Cura.

**Application version**
4.4.1

**Platform**
Windows 10

**Printer**
Ender 3 Pro

**Reproduction steps**
- Have pc on, audio running through bluetooth speaker
- Launch Cura
- Let Cura run for a minute or so
- Audio cuts out, speaker appears to turn off

**Actual results**
Audio cuts out after running Cura for a minute or so, bluetooth device turns off.

**Expected results**
Device should remian on and audio should continue playing while running Cura software

**Project file**
n/a

**Additional information**
- Windows 10
- There are multiple audio devices connected to this PC
- No interaction with Cura software is necessary (beyond launching it) to reproduce issue
- 100% reproduction rate

Notes: 
n/a

Ghostkeeper, 2020-02-04:
I don't use Bluetooth, so maybe it's only happening to the Bluetooth drivers, not the audio drivers.

The only thing I can think of is that this Bluetooth driver behaves as a serial input and Cura at some point tries to see if it's a printer, causing the drivers to crash. So maybe you can try going into the "Installed" tab of the Marketplace and disable the USB Printing plug-in? See if that resolves the problem?

graycatgrayhat, 2020-02-10:
**Ultimaker Cura Versions**
- 4.4.1
- 4.1.0
- 3.6.0

**Reproduction steps** _100% reproduction rate_

1. Start windows
2. Bluetooth speaker connected
3. Launch Cura     
4. Let run for a minute or so   
5. Audio cuts out, the speaker appears to turn off, stereo shows disconnected

**Motherboard**

- MSI X470 GAMING M7 AC (MS-7B77) (AM4)
- _Intel BlueTooth Driver Version 21.50.1.1_

**Operating System**

- OS Edition: Windows 10 Pro
- OSVersion: 1909
- OS Build: 18363.592

Ghostkeeper, 2020-02-12:
I did the exact same steps, though with a different motherboard and different drivers (and using headphones instead of speakers), but it doesn't get interrupted here.

I'll also refer to my previous question:

> The only thing I can think of is that this Bluetooth driver behaves as a serial input and Cura at some point tries to see if it's a printer, causing the drivers to crash. So maybe you can try going [...]

graycatgrayhat, 2020-02-12:
> I did the exact same steps, though with a different motherboard and different drivers (and using headphones instead of speakers), but it doesn't get interrupted here.
>
> I'll also refer to my previous [...]

This is the fix for now, I really appreciate the investigation and help!

Ghostkeeper, 2020-02-14:
That's a workaround for now. But the original issue stems from the issue discussed here then: https://github.com/Ultimaker/Cura/issues/5207

I'm closing this as a duplicate. That's a fairly popular issue, but hard to fix and not in Ultimaker's interest to spend time on.

mahtDFR, 2020-02-14:
Duplicate of #5207

closed 2020-02-14 by mahtDFR