Ultimaker/Cura #6837: Please implement the 3d connexion space navigator in Cura

ylectron, 2019-12-21:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

**Additional context**
(Add any other context or screenshots about the feature request here.)

adamloose, 2019-12-21:
It´s a good idea!

fieldOfView, 2019-12-21:
Duplicate of #1152

closed 2019-12-23 by Ghostkeeper