Ultimaker/Cura #6838: [4.4 and earlier] Inner Wall Speed in gcode does not match speed in the settings for layers without skin (ie top/bottom)

newtorah, 2019-12-21:
**Application version**
Tested with both 4.4 and 4.0

**Platform**
Windows 10, AMD 3600x, GTX970

**Printer**
Ender 3

**Reproduction steps**
Step 1: Open STL in Cura (I used this stl https://www.thingiverse.com/thing:1278865)
Step 2: select "Standard Quality 0.2mm"
Step 3: Change "Inner Wall Speed" to above 34mm/s (for example, I used 100mm/s)
Step 4: Slice
Step 5: Inspect Gcode for inner wall on a layer without skin/ top/bottom

**Screenshot(s)**
[image]

**Actual results**
Gcode for feedrate for inner wall layer is lower than actual expected value for layers without skin, G1 F4790.7

**Expected results**
Gcode for feedrate for layers with skin should be G1 F6000 (ie: 60*Inner Wall Speed)

Note: Other speed settings may have similar issues, this is just the one I noticed specifically, and spent the time to figure out

smartavionics, 2019-12-21:
It's almost certainly due to the minimum layer time setting slowing the print down to avoid overheating. The layers with skin take longer to print so the walls can be printed faster.

newtorah, 2019-12-21:
Yup, 100% was minimum layer time, didn't even think of that. Thanks!

closed 2019-12-21 by newtorah