Ultimaker/Cura #6839: [4.4.0]Unable to change color scheme in old machine

wjk199511140034, 2019-12-22:
**Application version**
4.2.1 4.3.0 4.4.0

**Platform**
WIN 10 x64
with intel E5500 + nvidia G210 or intel E5430 + nvidia G210

**Printer**
custom

**Reproduction steps**
Just after installed

**Screenshot(s)**
[image]

**Actual results**
I attempted to toggle preference>general>force layer view capability mode, but it won't make any difference, I also attempted to change cura version and driver version, no difference too.

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
[cura.log](https://github.com/Ultimaker/Cura/files/3992303/cura.log)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2019-12-22:
The geforce 210 does not support opengl 4.1, which is required for the other schemes. With that GPU, compatibility mode is the only available mode, regardless of the option in the preferences.

closed 2019-12-23 by nallath

pulidoow, 2021-04-09:
Good morning. I have the same problem, but my notebook has two video adapters (Intel HD Graphics 3000 and NVIDIA GeForce GT 525M). Can Cura show different color schemes using either of these two cards? How can I do to see this different schemes? I've already install the last OpenGL software for the NVIDIA adapter. Thanks for reading.

Ghostkeeper, 2021-04-20:
There's a handy reference application by which you can check whether your graphics card supports the 3D layer view in Cura. You can download it here: https://github.com/Ultimaker/Cura/releases . There's nothing quite like actually trying to see if it works, 's what I'm saying.

To display the 3D layer view (and use its features such as different colour schemes) Cura needs OpenGL 4.1, a standard from 2010. As far as I can see online, neither of those graphics cards supports that. But the information I find online may be outdated, of course.

erkserkserks, 2021-12-20:
It's possible to use a software OpenGL renderer like llvmpipe. On Windows, 
https://github.com/pal1000/mesa-dist-win

Unzip a release, e.g. mesa3d-21.3.2-release-mingw.7z, and run `perappdeploy.cmd`. It will ask where to install the software renderer to, point it to the Cura installation directory, e.g. `C:\Program Files\Ultimaker Cura 4.12.1`

My laptop's GPU (Intel HD Graphics 3000) only supports OpenGL 3.1, but I was able to use the software renderer to unlock all of Cura's rendering modes (layer thickness, speed, flow). Performance is pretty bad (~1 FPS), but it is at least possible to use these features.