Ultimaker/Cura #6841: Most text in cura has disappear

LuisLlana, 2019-12-22:
Hello,
  I have the same problem an in issue #5645, most of the test has disappeared.  I was running Ubuntu 18.04 when this problem first arose. I have not been using Cura for a time. Meanwhile, I ugraded to Ubuntu 19.10. The problem persists. 

**Application version**
 dpkg -l cura
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-====================================
ii  cura           3.3.1-2      all          GUI G-code generator for 3D printers

**Platform**
Ubuntu 9.10

00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor PCI Express x16 Controller (rev 06) (prog-if 00 [Normal decode])
        Flags: bus master, fast devsel, latency 0, IRQ 16
        Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
        Capabilities: [88] Subsystem: ASUSTeK Computer Inc. Xeon E3-1200 v3/4th Gen Core Processor PCI Express x16 Controller
        Capabilities: [80] Power Management version 3
        Capabilities: [90] MSI: Enable- Count=1/1 Maskable- 64bit-
        Capabilities: [a0] Express Root Port (Slot+), MSI 00
        Capabilities: [100] Virtual Channel
        Capabilities: [140] Root Complex Link
        Capabilities: [d94] #19
        Kernel driver in use: pcieport

**Reproduction steps**
I cannot say.

**Screenshot(s)**
[image]
[image]

Ghostkeeper, 2019-12-23:
Could you try moving or deleting the `~/.qtshadercache` folder (while Cura is not running)?

LuisLlana, 2019-12-23:
I did not have that directory. Instead, there were two similar direcories inside ~/.cache: qtshadercache and qtshadercache-x86_64-little_endian-lp64
I removed both without success. In fact, I have deleted the whole .cache directory.
I have other users in the same machine and all of them have the same problem. I have removed (with --purge option) and install again the package and it still does not work.

Ghostkeeper, 2019-12-23:
Does the problem also occur when you're using our AppImage releases?

If not, then it's some dependency coming from your system that renders these fonts differently. It could be FreeType, or perhaps Qt itself. Myself I'm running from source on Ubuntu 19.10 as well. I'm using `libfreetype6/eoan,now 2.9.1-4` to render the fonts, and Qt 5.13.2 via the PyQt5 Pip package.

canonex, 2021-01-07:
I confirm this behavior on Debian 10.
The version that comes with the Debian installer is the 3.3.1, quite old.
_An updated appimage (v. 4.x) has no issue._

Ghostkeeper, 2021-01-18:
We are not maintaining the Debian installer. Debian has a maintainer to release it to their package manager but they are not really involved in development with us. We only support the AppImage.

Though we're happy to fix issues the package maintainers are encountering as long as they're not in the way of our normal development, we don't have the resources to maintain releases on all those platforms out there. AppImage is our only Linux distribution.

onitake, 2021-01-20:
If #5645 was fixed by #7139 , then it's not in 4.8 yet, as far as I can see. You'll have to wait for the next Cura release.

And a workaround for the rendering issue is forcing the material Qt theme:

```shell
QT_QUICK_CONTROLS_STYLE=material cura
```

onitake, 2021-01-22:
Debian buster (stable) still carries Cura 3.3, which is severely outdated now.

If you want to try a newer version, please upgrade to bullseye (testing) or wait until it becomes stable later this year.
Or rely on the AppImage version provided by UltiMaker for the time being.

GregValiant, 2024-11-22:
I'm cleaning house.
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

onitake, 2024-11-23:
To my knowledge, this has been working fine since at least Cura 5.0.
I think the issue can be closed.

closed 2024-11-23 by fieldOfView