Ultimaker/Cura #6842: [4.4.0]Tree support not generated for overhang

MisterScott, 2019-12-23:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

MisterScott, 2019-12-23:
Works when I adjust Tree Support Branch Diameter Angle from 15 down to 10.  I think this supports the non-intersecting-cones hypothesis.

[image]

Ghostkeeper, 2019-12-23:
I'm not able to reproduce this problem. Here I have a bit of overhang, and clearly it's generating tree support:
[image]

Please fill in the issue template with your reproduction steps and observed behaviour. You seem to have had some trouble editing that so maybe it didn't publish it in the end?

Ghostkeeper, 2019-12-23:
Note that the default for Branch Diameter Angle is 5 degrees. I tried with 15 as well and it seemed to work fine. But of course there will be values for this angle where it produces unprintable results, like 89.999 degrees and such.

closed 2020-06-19 by no-response[bot]