Ultimaker/Cura #6843: [master] crash when loading project file

smartavionics, 2019-12-23:
**Application version**
master

**Platform**
Linux

**Printer**
Project being loaded was for a custom printer.

**Reproduction steps**
I attempted to load a project file sent to me by someone else.

**Screenshot(s)**
No.

**Actual results**
Cura crashed with the following diagnostics...
````
2019-12-22 23:19:53,167 - INFO - [MainThread] 3MFReader.ThreeMFWorkspaceReader._processQualityChanges [745]: Loading custom profile [PLA on glass for pipe 
fittings] from project file
2019-12-22 23:19:53,197 - DEBUG - [MainThread] 3MFReader.ThreeMFWorkspaceReader._processQualityChanges [772]: Created new quality changes container [custom_pla_on_glass_for_pipe_fittings]
2019-12-22 23:19:53,269 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [58]: An uncaught error has occurred!
2019-12-22 23:19:53,271 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [61]: Traceback (most recent call last):
2019-12-22 23:19:53,271 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [61]:   File "/opt/cura/lib/python3/dist-packages/cura/CuraApplication.py", line 1003, in event
2019-12-22 23:19:53,271 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [61]:     return super().event(event)
2019-12-22 23:19:53,272 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [61]:   File "/opt/cura/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 429, in event
2019-12-22 23:19:53,272 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [61]:     event._function_event.call()
2019-12-22 23:19:53,272 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [61]:   File "/opt/cura/lib/python3/dist-packages/UM/Event.py", line 189, in call
2019-12-22 23:19:53,272 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [61]:     self._function(*self._args, **self._kwargs)
2019-12-22 23:19:53,273 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [61]:   File "/opt/cura/lib/python3/dist-packages/cura/Utils/Threading.py", line 34, in _handle_call
2019-12-22 23:19:53,273 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [61]:     ico.result = func(*args, **kwargs)
2019-12-22 23:19:53,273 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [61]:   File "/opt/cura/lib/cura/plugins/3MFReader/ThreeMFWorkspaceReader.py", line 713, in read
2019-12-22 23:19:53,273 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [61]:     self._processQualityChanges(global_stack)
2019-12-22 23:19:53,273 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [61]:   File "/opt/cura/lib/cura/plugins/3MFReader/ThreeMFWorkspaceReader.py", line 797, in _processQualityChanges
2019-12-22 23:19:53,274 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [61]:     intent_category = quality_changes_intent_category_per_extruder["0"]
2019-12-22 23:19:53,274 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [61]: KeyError: '0'
````

**Expected results**
Normality.

**Project file**
The project file is available at https://community.ultimaker.com/topic/30600-odd-infill-issue

**Log file**
See above snippet.

**Additional information**
This patch stops the crash and allows the project to load...
```
diff --git a/plugins/3MFReader/ThreeMFWorkspaceReader.py b/plugins/3MFReader/ThreeMFWorkspaceReader.py
index dd35484c3..e25cf4882 100755
--- a/plugins/3MFReader/ThreeMFWorkspaceReader.py
+++ b/plugins/3MFReader/ThreeMFWorkspaceReader.py
@@ -794,7 +794,10 @@ class ThreeMFWorkspaceReader(WorkspaceReader):
                 if not global_stack.extruders:
                     ExtruderManager.getInstance().fixSingleExtrusionMachineExtruderDefinition(global_stack)
                 extruder_stack = global_stack.extruders["0"]
-                intent_category = quality_changes_intent_category_per_extruder["0"]
+                try:
+                    intent_category = quality_changes_intent_category_per_extruder["0"]
+                except KeyError:
+                    intent_category = None
 
                 container = self._createNewQualityChanges(quality_changes_quality_type, intent_category, quality_changes_name, global_stack, extruder_stack)
                 container_info.container = container
```

Ghostkeeper, 2019-12-23:
Indeed, this bug reproduces.

We might need to fill that dictionary with `"0": "empty"` as it's also being used elsewhere.

LipuFei, 2019-12-24:
JIRA ticket: CURA-7078

evtrados, 2020-01-03:
The issue has been resolved.

closed 2020-01-03 by evtrados