Ultimaker/Cura #6844: [4.4.0] Unable to set specific nozzle size (0.25, 0.35)

texastoast, 2019-12-23:
**Application version**
4.4.0

**Platform**
Windows

**Printer**
Creality CR-10S

**Reproduction steps**
1) Open Cura
2) Settings -> Extruder 1 -> Nozzle Size
3) Nozzle sizes listed are all rounded to the nearest tenth.

- or -

1) Settings -> Printer -> Manage Printer
2) Select/Activate your printer
3) Click Machine Settings
4) Click Extruder 1
5) Nozzle diameter field is missing

**Screenshot(s)**
https://imgur.com/a/hx6x5M1

**Actual results**
Nozzle sizes are all preconfigured and rounded to nearest tenth. Nozzle diameter setting on extruder is missing.

**Expected results**
I should be able to either select my nozzle size (0.25 or 0.35) or I should be able to enter my specific nozzle size.

**Project file**
project file not required to reproduce

**Log file**
No log

**Additional information**
These nozzles are popular and are sold by E3D, and they come in 0.15, 0.25, 0.35 sizes. 
https://e3d-online.com/nozzle-fun-pack

texastoast, 2019-12-23:
After playing around with the settings and seeing what changed, I saw that the nozzle size is the same as the line width, and that changing line width to equal the new nozzle size updates everything that needs updating in the profile.

This was not obvious, nor was it terribly easy to figure out. Other slicers set the line width to something larger than the nozzle size. Apparently Cura handles this differently.

I'm closing this report since I figured it out and it isn't a bug. I suggest a better interface for this since the ability to select nozzle size is so predominant, it led me to believe I needed to somehow add a 0.25 option to the dropdown.

closed 2019-12-23 by texastoast

Ghostkeeper, 2019-12-24:
The set of Creality CR-10s profiles pre-defines the available nozzles for that printer. If Cura gets a set of pre-defined nozzles, then it won't allow adjusting the nozzle size from the interface (unless the Printer Settings plug-in is installed from the Marketplace). It is considered a machine setting since adjusting the nozzle size is a hardware change just as much as swapping out the feeder or changing the build plate heating element.

The line width can be different from the nozzle size, and some profiles make it bigger than the nozzle size like in those slicers you've been using. In fact, for your printer it used to be set to 110% the nozzle size. Since 4.3 the Creality community changed it back to be equal to the nozzle size because [it was found to print better that way](https://github.com/Ultimaker/Cura/issues/6106).