Ultimaker/Cura #6846: [Ver. 4.4.0 & 4.4.1] Locks up Ender 3 Printer (Crashes)

NinjaDanny87, 2019-12-23:
**Application version**
4.4.0 & 4.4.1

**Platform**
Windows 10 Pro, 
Nvidia Driver Ver. 441.66

**Printer**
Creality Ender 3 (From Original Manufacturer)

**Reproduction steps**
**1.** Started Cura, plugged in printer via USB, waited and got no response. Printer locked up and would not respond to commands via turn dial, Cura did not recognize the printer.
**2.** Repeated the same process with pronterface and was able to connect to the printer via COM7.
**3.** Updated Marlin Firmware, and repeated steps 1 & 2. Printer failed to connect to Cura.

**Actual results**
Print was only able to connect to pronterface when connected via USB, failed to connect with the newer versions of Cura. Never had an issue with version 4.2.1
When the printer freezes up after connection to Cura via USB, the screen on the printer freezes and controls do not respond. Also with Cura Ver 4.4.0 the Z axis stepper motor made a "Wirring" sound like a police siren. The results are the same with Cura Ver. 4.4.1 except there is no siren sound being produced.

**Expected results**
The Cura "Monitor" Tab should have loaded the printer controls upon successful connection. 
The Monitor tab only shows a blank screen when no printer is connected, and will not connect to the printer anymore. No issues prior to this update.

Any Ideas why this is the case? I do not think this issue with with the printer, as it works with other slicers just fine. Also I was using Cura exclusively up until this recent update, now I can't connect to print. Using an SD card for the time being, but I much rather use the USB connection when printing.

rburema, 2019-12-24:
Hi @NinjaDanny87 
What happens with 4.3.0? That could narrow down the potential changes a bit.

printerpietje, 2019-12-24:
This  4.4.1 version does not starts in fusion 360 wen i click on 3d print, select the making and  chose cura and click ok.
Please help.

rburema, 2019-12-24:
Hi @printerpietje 
 - We've noted before (see [here](https://github.com/Ultimaker/Cura/issues/6697)). that's something the Fusion360 devs will have to change in their setup, we can't do anything about it.
 - Please don't post in unrelated issues, but make a new topic.

NinjaDanny87, 2019-12-25:
> Hi @NinjaDanny87
> What happens with 4.3.0? That could narrow down the potential changes a bit.
It seems to work just fine. 4.3.0 works with USB.

I am using 4.4.1 normally though, I just can't connect to the print though the USB port, and it seems like all versions run normally for regular editing etc. 

_On a side note could be opened for a separate topic, I've noticed there are no profiles to select as well. Odd..._

Ghostkeeper, 2019-12-30:
The only commits to the USB printing plug-in between 4.3 and now are:
* https://github.com/Ultimaker/Cura/commit/c875252b53443d18ccd14dbe747e8561fb58c637#diff-a0da3e9c4706ebfbfaa54c9c2db29ef8 (whitespace change, shouldn't do anything).
* https://github.com/Ultimaker/Cura/commit/c62ff262ed3dbdb70e288086c1c5603d038ff181#diff-a0da3e9c4706ebfbfaa54c9c2db29ef8 (version increment, doesn't actually change anything).
* https://github.com/Ultimaker/Cura/commit/c2e6116983ba23c215ecdd4a90f0f37b9b617449#diff-a0da3e9c4706ebfbfaa54c9c2db29ef8 (adds a name to the thread to make things easier to debug, shouldn't change anything again).

We didn't change the major dependency of that plug-in PySerial either, it's still on version 3.4 (last updated with Cura 4.1.

So I have no idea what could've broken this.

As far as I'm concerned, there should be no serial communication that freezes firmware, so we should consider reporting this to Creality as well. If Cura does something wrong we should fix it of course, but we'll need to know what Cura is doing wrong then. I don't have an Ender 3 to test this with though.

DJDC13, 2020-02-16:
I am now having a similar issue. 4.4.1 was working fine for me but now every time Cura is opened the printer freezes and the screen control does not work. Com 6 is showing as connected on my system as normal. If I do not open cura but plug the USB in it continues working it is only when Cura opens it freezes.

closed 2020-06-19 by no-response[bot]