Ultimaker/Cura #6848: Feature blowing mode for hanging parts

Goodfeat, 2019-12-24:
Hello. 

Is it possible to add a function for blowinghanging parts.

Some parts can be printed without supports. However, in long sections of overhanging parts without supports, the plastic is twisted upward.

The program has a similar setting

[image]

But this setting only works on existing supports.
Is it possible to add a function for blowing hanging parts without support?

Thanks

Ghostkeeper, 2019-12-24:
There is one, sort of. It's the [Bridging Fan Speed setting](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/experimental/bridge_fan_speed.md). It's under Experimental, and can be seen once the extra bridging settings are enabled.

Goodfeat, 2019-12-24:
Well, I will try it if I find it.
I just don’t use bridge setup

Goodfeat, 2019-12-24:
Unfortunately this is not quite what I need
This setting only works at extreme angles.
when plastic is in the air
[image]

The program has two more settings

[image]

These settings are clearly related to the dangling elements.
Is it possible to add here the third setting for blowing the hanging elements?

smartavionics, 2019-12-24:
If you install one of my Cura builds you get these settings...

[image]

You can find my builds at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. Please read the README.md there.

Ghostkeeper, 2019-12-24:
Ah right, this is a duplicate of https://github.com/Ultimaker/Cura/issues/6656 then. Please continue the discussion there!

closed 2019-12-24 by Ghostkeeper