Ultimaker/Cura #6849: Can some explain the profile file structure for me?

rickyelqasem, 2019-12-24:
I have an idea to build a useful opensource  tool. I know how to edit them but I don't understand why there is multiple files.  I've seen the profile structure pic but I cannot decipher the relation in the files. Thanks

**Application version**
4.4

**Platform**
Windows

**Printer**
Ender 3

**Reproduction steps**
NA
**Screenshot(s)**
NA

**Actual results**
NA

**Expected results**
NA

**Project file**
NA

**Log file**
NA

**Additional information**
(Extra information relevant to the issue.)

nallath, 2019-12-24:
https://github.com/Ultimaker/Cura/wiki/Profiles-&-Settings <- Is this what you are looking for?

rickyelqasem, 2019-12-24:
@nallath thx, will look now.

closed 2019-12-30 by Ghostkeeper