Ultimaker/Cura #6852: Fill top surface and Ironing areas using the same direction.

printingotb, 2019-12-24:
**Is your feature request related to a problem? Please describe.**
Areas on top layers, that are filled from two different directions create a visual effect that is not wanted. See attached pictures, changing viewing angle, drastically changes the look. This is especially true on metallic filaments like Easyprint gold PLA that I used here. 

**Describe the solution you'd like**
I would like to have a option to fill top surface layers and ironing layers using the same fill direction no matter if this creates longer travel paths or not. 
Being able to choose what direction is not important, the important thing is that areas aren't filled from two directions and meet in the middle. As this affects the reflection of the surface and how visible the "seam" where areas meet will become. I have noticed that the seam is more pronounced when areas are filled from two directions and meet in the middle, than areas where the nozzle goes back and keeps filling the same area in the same direction. 

**Describe alternatives you've considered**
I have made a lot of tests using different flow rates to make sure that the issue isn't caused by over or under-extruding. I learned that the "seam" where areas meet can be reduced some by deliberately under-extruding the top surface skin layer by 2-4%. This also helps ironing cover it up better. 

**Affected users and/or printers**
Primarily, anyone using Ironing will benefit the most from this. Users using top surface layer settings for a smoother surface will also benefit. Normal top layers are as far as I know typically not used to create very smooth surfaces. 

**Additional context**
[image]
[image]
[image]

Ghostkeeper, 2020-01-02:
I know the issue you're facing. I've seen it before as well.

I've discussed this with the team, but we've decided that this doesn't get our priority right now. We'll defer this for later when we have more free time or if the importance of this changes somehow.

printingotb, 2020-01-02:
I understand, thank you!

luxology, 2020-01-05:
This is important problem for who making business card type text based surfaces. Thanks Ghostkeeper for your effort.

jasoncoon, 2020-10-17:
It looks like this is being implemented in Prusa Slicer 2.3 as "monotonic infill":

https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-alpha1
https://twitter.com/TheZuza/status/1315735630429802504
https://twitter.com/Greg191134/status/1274075292982788096

MrEd88, 2020-11-14:
Please check this and consider its application, this not only helps with the surface shape but may also reduce long travel movements and scratches.

GoremanX, 2020-12-31:
Oh, so it's not a defect on my printers 🤣 This has been driving me nuts for months, it's literally the only defect in any of my prints

o-nix, 2021-01-02:
Yes, please.

fvrmr, 2021-02-04:
I will close this issue, because it is a duplicate of #8909

closed 2021-02-04 by fvrmr

o-nix, 2021-02-04:
@fvrmr the issue you referenced is about **infill**, not **ironing**, are you sure about closing this one?

GoremanX, 2021-02-04:
It's the same issue with an incorrect word used.

I'm more confused about closing this issue rather than the newer duplicate. Honestly it looks like "hey someone else's software does this, we should copy it". Whereas this issue predates the feature in the competing software and does a much better job of describing the problem.

fvrmr, 2021-02-05:
On the newer issue we've created a ticket for in our backlog. So we have an overview about the feature request, also now it is connected to this issue. 
Hope that makes sense.