Ultimaker/Cura #6853: [4.4.1] Cannot recover online backup on MacOS

sezginriggs, 2019-12-24:
**Application version**
4.4.1 (also occurs on 4.4.0 too)

**Platform**
MacOS Catalina v10.15.2

**Printer**
Creality Ender-3

**Reproduction steps**
I have Cura 4.4.1 on Windows 8.1 machine and taking online backup and restoring it on MacOS (with Cura 4.4.1)

Restore function asks for restarting the Cura, but unfortunately it doesn't restart the app. It downloads backup file to /Users/<username>/Library/Application Support/cura but doesn't extract it. 

Also I tried to remove previous profile (4.4 folder) and extract it myself but after doing and starting Cura, Cura still opens the wizard. On wizard I accepted agreements and followed steps until "Add a printer" screen. Restarted Cura here, and the wizard opened again... Completed wizard this time but unfortunately it didn't load any custom profiles or any other settings from backup.

After this, I tried to do "restore" again, restore steps required to restart the app, didn't restart it, but opening the app again myself recovered custom profile names but "Build plate temperature" was right when "printing temperature" is different from the backed up settings.

ChrisTerBeke, 2019-12-24:
Can you share your Cura log files with us?

sezginriggs, 2019-12-24:
[cura.log](https://github.com/Ultimaker/Cura/files/3998929/cura.log)

@ChrisTerBeke I hope this will help. This is the log file of the computer I try to do the restore.

Ghostkeeper, 2019-12-30:
> but opening the app again myself recovered custom profile names

So something did get restored from the back-up then? Maybe we're looking at a version upgrade problem. What version was the back-up created from?

Ghostkeeper, 2019-12-30:
The log file in this case doesn't indicate much of what is going wrong. The back-up was restored before the run that the log file was logging. It's just showing that you were slicing a couple of times on an Ender 3, basically.

sezginriggs, 2019-12-31:
@Ghostkeeper back-up created from 4.4.1 and restored on 4.4.1 too. Yes, profile name etc. restored after manually restarting the app.

And for the log, yes i made some slicing but right after restoring, maybe in a hour...

Ghostkeeper, 2020-12-04:
We had this on our backlog for a while, but it was just deferred by our PO due to shifting priorities. We'll keep the ticket open but realistically we won't be giving this any attention soon.

closed 2023-06-24 by github-actions[bot]