Ultimaker/Cura #6854: [4.4.1] Crashing on menus

ruasablub, 2019-12-25:
Cura version 4.4.1

Windows 10 laptop, Nvidia GTX 1050 Ti

Printer: CR-10 Mini

What should happen: I open a menu, whether from the toolbar at the top or somewhere else (filament selection and such), select the option I want, and go about my day

What actually happens:  If the Cura window is on my laptop screen instead of my external monitor, Cura freezes and stops responding
[cura.log](https://github.com/Ultimaker/Cura/files/3999245/cura.log)

rburema, 2019-12-29:
Hi @ruasablub 
How is your external monitor connected to the laptop? (Do you use a dock for example?) Are the GPU drivers up to date?

Ghostkeeper, 2019-12-30:
Unable to reproduce this so far, neither on Ubuntu 19.10 on my work laptop (Intel HD graphics integrated GPU thingy), nor on my home desktop computer with Windows 8.1, Nvidia GTX 2070. It works fine on both screens on either computer.

TheDroneDoctor, 2019-12-30:
I am running an MSI with a 2060 rtx. I use a 4k TCL TV as a monitor via HDMI.  Cura 4.4.0 worked fine last week when connected.  I updated to 4.4.1 and now when I am connected to the monitor via HDMI Cura won't run.  It gets past the loading menus and then stops responding.  If I unplug the hdmi and start Cura it works fine.  If I start Cura then plug in the HDMI then Cura crashes.

fieldOfView, 2019-12-31:
On the "General" pane of the preferences, there's an option "Restore window position on start" that is checked by default. Could you uncheck it and then restart Cura? Does that change anything?

closed 2020-06-24 by no-response[bot]