Ultimaker/Cura #6855: Cura 4.4.1 STILL crashing ENTIRE Computer with Ryzen Processors

verfassungsschutzz, 2019-12-25:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

rburema, 2019-12-29:
Hi @verfassungsschutzz 

Can you confirm or deny:
 - Ryzen 2700/3700
 - Asus motherboard
 - while/involving USB printing

verfassungsschutzz, 2019-12-30:
Ryzen 5 3600x
Asus Primex470 Pro
Not USB Printing. Just having the Programm open.

Ghostkeeper, 2019-12-30:
Duplicate of https://github.com/Ultimaker/Cura/issues/5265.

It's not fixed because we don't know how any Python code could ever cause a CPU to lock up. Other applications are having the same problem. Seems to be something to do with the PSU when the CPU instructs it to go into C6/C7 state and the motherboard claims it can do that but doesn't. Please continue discussion over at #5265.

closed 2019-12-30 by Ghostkeeper

Razorlights, 2020-06-27:
Hi all, know this is a relatively old forum and maybe this has been resolved already, but  I have found something that works for me!

Ryzen 2600X
ASUS X470 Prime-Pro
GTX 1660 Super
16GB DDR4 3000Mhz
Windows 10 Home
Cura 15.04.3 (Win 8 compatibility)

So I was reading about people's thoughts on the CPU load and disabling C state, this is not quite an option for Ryzen CPU's.
However there is something similar that works for me, I enabled all 4 cores in the BIOS and not crashed so far over 48 hours of use over a few weeks, when usually I had the same problem every hour or so, however I am using an older stable version (in my opinion) of Cura and using windows 8 compatibility, so maybe all these factors together are a win for my system. Hope this helps someone!