Ultimaker/Cura #6856: Y Layer Shift on CR10-S500, 4.4.0

ackerman76, 2019-12-25:
**Application version**
4.4.0

**Platform**
Win 10 64

**Printer**
CR10-S500

**Reproduction steps**
1. Printed for one year, no issues. I've never seen any shift before. The only thing that changed is I upgraded from Cura from 3.6 to 4.4.
2. Printed using the same profiles I have been using all along.

**Screenshot(s)**
[image]
[image]

**Actual results**
Y Layer shift. Model upper layers extend toward front of printer (bed and print is shifted to rear of printer)

**Expected results**
No Layer shift

**Project file**

**Log file**

**Additional information**
Accel control, Jerk and Z Hop all DISABLED from the start. No pulleys or mechanical issues, nothing loose. Disabling these are the only suggestions I found and them already disabled. Should I try turning accel control on?

I did notice that my existing profiles got random settings changed, for example the temp control was 100 less exactly. This could be a bug or could be focus / freezing of cura with user input (which I assumed).

The movements in-layer are different and quite extreme. For example for a thin shell, the slicer is sending the head all the way around the nearest circumference after doing a tiny bit of the middle bottom/top fill. Over and Over. Ridiculously inefficient!

It could be there are new movements programed in 4.4 which exceed my Y stepper current. I would have to tear my box apart to measure the driver voltages.

Sad to ruin prints after 24+ hours! So wasteful! I am going back to 3.6 :(

smartavionics, 2019-12-25:
Check the start gcode in the machine settings dialog, perhaps the accel/jerk has been set too high there.

Ghostkeeper, 2019-12-30:
> I did notice that my existing profiles got random settings changed, for example the temp control was 100 less exactly.

That's to be expected, as we regularly get updates from the Creality community for the default profiles. In particular there was a big update at Cura 4.1 that changed many things around. I wouldn't expect it to have a 100 degree drop in printing temperature though. Wouldn't that bring the printing temperature of PLA to like 90 degrees?

> The movements in-layer are different and quite extreme. For example for a thin shell, the slicer is sending the head all the way around the nearest circumference after doing a tiny bit of the middle bottom/top fill. Over and Over. Ridiculously inefficient!

Can you show us an example of this happening? A project file would help.

> With the exact same profile in 3.6, the print will be 1.5h quicker.

That's also what we saw in the 4.1 update from just slicing (we can't print without having the actual printers). From what our users reported the printing quality is increased too however so you're better off picking a greater layer height. In the end the print quality per duration seemed to have increased.

> Also looking at how the printer moves, the travel accelerations or speeds are out of control -- WAY TOO FAST. This is strange because it is all disabled according to the profile. I suspect the UI is lying or there are other settings that do not obey limits.

Indeed I think it'd be better if the profile just enabled acceleration control and set the acceleration to 500 that way rather than disabling it and then setting it via the start g-code. That would look better for the users of that profile. Can you verify that the behaviour is the same, if you change those?

> I also believe there are profile bugs moving from old to new versions of cura. Minimum distance exclusion window is 3 mm in 3.6 (matches Retraction distance), but the exact same profile shows 10 mm when loaded in 4.4. Who knows what else is going on that I cannot see.

Again, that's to be expected, not a bug. We update the default print settings with what the Creality community provides us. Those defaults are used for any setting that you didn't modify in your own profiles. Between 3.6 and 4.4 dozens of settings have changed. We also change the behaviour of the slicer once in a while so that minor things could be different even if all the settings were the same. If you really need the g-code to be 100% the same, then don't update anything (or save the g-code locally). You can run multiple versions of Cura alongside each other if you need the new features for new prints.

closed 2024-12-10 by github-actions[bot]