Ultimaker/Cura #6858: Support generation creates incomplete support towers, large air gaps in support

3DTanny, 2019-12-26:
**Application version**
4.4.1
**Platform**
Windows 10 Pro
**Printer**
Multiple - Issue has been present for a long time (since version 3.x at least)
**Reproduction steps**
1 - Enable "Generate Supports"
2 - Load model, usually with organic shapes
3 - Witness dumb supports in preview

**Screenshot(s)**
I have some of a model I was recently working on, they are NSFW. Can be provided upon request.

**Actual results**
Supports generated that stop at seemingly random layers and do not support anything. Other supports may stop at some random layer, then have an air gap of several dozen layers, then start again in mid-air.

**Expected results**
Supports go all the way to the spot they are supposed to be supporting.

**Project file**
WARNING - NSFW model. Only reason this particular model is the example is because this model is one I am asked to print a lot and the support generation makes it impossible to slice in Cura. There are other models this happens on, but I can't remember which ones.
https://www.dropbox.com/s/psvqbstuh8gpppq/ASX1_Dino_Dick_V2.3mf?dl=0

**Additional information**
I will start looking through my file catalogue for other files that have this issue. This happens no matter which printer I am using. I have even tried to slice them for built-in Ultimaker printers with default profiles and this happens. Please fix this, it has been happening since at least version 3.4 that I can easily recall.

nallath, 2019-12-30:
I can't really load your project since I don't have the sidewinder definition. Could you point me to where you got those profiles from?

I've tried slicing the model with the default settings from Ultimaker and I don't see any of the weird behavior that you describe, so my best bet is that it's some default setting that was set by whoever created the machine definition you are using.

Ghostkeeper, 2019-12-30:
Try reducing the Minimum Support Area setting to 0. Does that solve things?

3DTanny, 2019-12-30:
Okay, so after Ghostkeeper's comment, I turned the minimum area down to 0 and it revealed the true bug. After turning it down to 0, supports went all the way to the supported area and came to a point. Then I turned "Enable Support Towers" off and put the minimum support area back to my default, and the partial support generation no longer happened. 

The true bug is:
Cura partially generates supports for areas below the minimum threshold when Support Towers are enabled.

Ghostkeeper, 2019-12-31:
The way that the Minimum Support Area setting currently works is to remove areas smaller than a certain area after all of the support areas are created. It's one of the last steps. I think in general we should consider removing that or revamping how it works because in many cases it breaks things. It doesn't jive well with support towers at all.

Ghostkeeper, 2020-06-29:
Still a problem.

tizianoj, 2020-12-08:
Happens to me too: if towers are on very thin areas, the tip of the tower gets truncated. In one case (the edge of a blade with some high support angle) I get it truncated also with this param set to 0, but it is a very special case