Ultimaker/Cura #6860: NOZZLE WIDTH DOES NOT CHANGE

BunnyTJ, 2019-12-27:
**Application version**
4.4.1

**Printer**
Tevo Tornado

In my previous 3.9 of CURA version, I tried changing my Nozzle diameter. It happened after multiple tries. Now after installing 4.4.1 it has the same problem. In settings the nozzle diameter does not change you type it in but as soon as you change screens it jumps back to 0.4

Twigho, 2019-12-28:
Yes me too, nozzle size for Custom FFFprinter is always reset at 0.4mm.. Same with anetA6. But with cr10 we can custom the nozzle...

fieldOfView, 2019-12-28:
Could you try using a comma instead of a period? eg 0,6 instead of 0.6

Twigho, 2019-12-28:
humm, yes that work now. I fought with during a long time this morning... (0,6 automatique change with 0.6 or now just .6 work too, without the 0 )
Thanks

rburema, 2019-12-29:
This seems like another duplicate of #6685 please continue the discussion there.

closed 2019-12-29 by rburema