Ultimaker/Cura #6861: Unable to slice

gazdela, 2019-12-27:
**Application version**
4.4.1

**Platform**
Windows 10

**Printer**
Ender 3

**Reproduction steps**
I change the printer, profile, and everything but it says "Unable to slice", yesterday I sliced and when I clicked again there was the same error

**Screenshot(s)**
[image]

**Project file**
[https://we.tl/t-f3iQBcHf91](url)

smartavionics, 2019-12-27:
Hi, that link to the project file doesn't work. Can you please zip the .3mf file and attach to this issue. Thanks.

MarkMan0, 2019-12-29:
I have the same problem. Sometimes I can't slice the same file, with the same settings.
What I did:

1. Add STL file
2. Slice with 0.2 layer height - success
3. Slice with 0.3 LH - success
4. Change LH to 0.2 again - slicing failed.

Nothing else was changed, just the layer height.
[file.zip](https://github.com/Ultimaker/Cura/files/4008175/file.zip)

nallath, 2019-12-30:
The project that you provided worked just fine for me, so I guess it's not something that is caused by the project. Could you share your log files with me? There might be something in there.

Ghostkeeper, 2019-12-31:
I've had this (or a similar) problem recently as well, with a very heavy model. It wouldn't start a slice. Changing anything would cause it to reslice and then it worked. I think something is nondeterministic.

Nothing could be seen in the log about it: CuraEngine was started and waiting for a slice to arrive. Cura front end finished error checking. I didn't think too much about it since a reslice worked for me.

javl, 2020-02-25:
I seem to have a similar problem with Cura 4.4.1 on the latest Debian (using the appimage).

Trying to slice on Standard Quality (or any setting for that matter) I often get a "Unable to slice" error when I've manipulated the model (mostly rotate / mirror) with no clear reason. 

For example:
(each step is done from a new, empty project and with the same model):

1. Import model and slice: no problem
2. import model, mirror on x axis and slice: "unable to slice"
3. import model, mirror on x axis, mirror again (so it is back to its original orientation, so with the same model and orientation as in step 1): "unable to slice"

The problem here is that I have no idea what causes the "unable to slice" error and I have no idea how to figure out what is wrong as I can't seem to find any logs for Cura? I read they are supposed to be in `~/.config/cura/4.4/` but this directory only has `cura.cfg`, `plugins.json` and empty `scripts` directory inside.

Attached is the .3mf file (renamed to .zip so I can upload it.)
[CE3_kap.zip](https://github.com/Ultimaker/Cura/files/4250036/CE3_kap.zip)

Ghostkeeper, 2020-02-28:
This may be a duplicate of #6766? At least for @javl it is, but for the original post it might not have been.

alexwoo1900, 2020-04-21:
I got this error too.
Can't slice model after mirroring it.

How to run:
From the source

Application version:
4.4.1 and 4.5.0

Platform:
Mac and Windows 7

Detail:
TypeError: Cannot cast array data from dtype('float32') to dtype('int64') according to the rule 'safe'

alexwoo1900, 2020-04-28:
**SOLVED**
For my question, the key is "no numpy-stl installed".(Numpy STL is not included in the numpy package)

**HOW DID I FIND OUT**
Error message "Cannot cast array ... rule 'safe'" misled my investigation.
The real clue is a piece of unimportant **WARNING** message "... Could not find numpy-stl. fall back to slower code."
That would make the stlreader to load a model with the legacy code.
I didn't go further but I believe that the old loading code is incompatible with the current model operation.

**SOLUTION**
pip install numpy-stl

**SUGGESTION**
The user need to be told that numpy-stl and trimesh are very necessary for running Cura perfectly.

closed 2020-06-19 by no-response[bot]