Ultimaker/Cura #6862: 4.4 Failed importing of old profile from 4.3

MunchHansen, 2019-12-27:
**Application version**
4.4.0

**Platform**
Mac OS X Mojave 10.14.6

Printer
Ender 3

Reproduction steps
Import the following profile:
https://we.tl/t-a1lRX52OnO 

(The link will be deleted in a week, ask me for new upload. :-) )

Screenshot(s)
No screenshot but here is the error text:

Failed to import profile from /Users/Frederik/Desktop/PLA_Ender3.curaprofile: Could not find a quality type normal for the current configuration.
The profile is not imported

Expected results
The profile should be properly imported even though its for an older version of Cura

Project file
N/A

Log file
2019-12-28 00:17:07,627 - DEBUG - [MainThread] cura.Settings.CuraContainerRegistry.importProfile [175]: Attempting to import profile /Users/Frederik/Desktop/PLA_Ender3.curaprofile

Additional information
This profile is for an older version of Cura, I think maybe 4.3.

sebachu, 2019-12-28:
Hi, I have the exact same bug when I tried to update from 4.3 to 4.4.

MunchHansen, 2019-12-28:
> Hi, I have the exact same bug when I tried to update from 4.3 to 4.4.

Did you solved this problem?

FlailAway, 2019-12-29:
Hi, is anyone looking at this issue? 
Can you make it ignore unknown items in an imported profile?
I have just had the same problem with **_"could not find a quality type super for the current configuration."_**

rburema, 2019-12-29:
Does this still happen with 4.4.1? We fixed a few of those in the patch.

MunchHansen, 2019-12-29:
> Does this still happen with 4.4.1? We fixed a few of those in the patch.

Hi there,

Unfortunately it still happen in 4.4.1. I have tried to export one cura profil and import it again to a new printer - getting the error as explained above in the topic. :-(

FlailAway, 2019-12-29:
> Does this still happen with 4.4.1? We fixed a few of those in the patch.

Yup, just grabbed 4.4.1 also in 4.3.0

sebachu, 2019-12-31:
> > Hi, I have the exact same bug when I tried to update from 4.3 to 4.4.
> 
> Did you solved this problem?

No, still an issue. I'm staying with 4.3 for now until it's fixed. I'm happy to post logs etc but the error is exactly the same as you posted originally above - logs show Cura could not find a qualiy type normal.

sebachu, 2019-12-31:
seems to be the same issue as this one:
"Can't import legacy profile: Could not find a quality type "normal" #6682"

sebachu, 2019-12-31:
@Ghostkeeper

Ghostkeeper, 2019-12-31:
Yeah, looks like a duplicate of #6682 then.

closed 2019-12-31 by Ghostkeeper