Ultimaker/Cura #6863: [4.4.0] Retracting twice in dual head prints

StephanRichter, 2019-12-27:
**Application version**
4.4.0

**Platform**
Linux Mint 4.15.0-72-generic #81-Ubuntu SMP Tue Nov 26 12:20:02 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

**Printer**
Ultimaker 2 upgraded to two Olsson-Blocks for 2.85mm material

Start G-Code: empty

End G-Code:
```
;Version _2.6 of the firmware can abort the print too early if the file ends
;too soon. However if the file hasn't ended yet because there are comments at
;the end of the file, it won't abort yet. Therefore we have to put at least 512
;bytes at the end of the g-code so that the file is not yet finished by the
;time that the motion planner gets flushed. With firmware version _3.3 this
;should be fixed, so this comment wouldn't be necessary any more. Now we have
;to pad this text to make precisely 512 bytes.
```

Extruder 1 / Start G-Code:
```
; ====== Enable Extruder ======
```

Extruder 1 / End G-Code:
```
; ====== Disable Extruder 1 ======
```

Extruder 2 / Start G-Code:
```
; ====== Enable Extruder 2 ======
```

Extruder 2 / End G-Code:
```
; ====== Disable Extruder 2 ======
```

**Reproduction steps**
1. add complementary objects to print with two colours/materials
2. slice with or without prime tower
3. Look at g-code

**Actual results**

On each material change, both the current and the new material are retracted.

Generated G-Code (dash-intended comments added afterwards):

```
;FLAVOR:UltiGCode
;TIME:427
;MATERIAL:294
;MATERIAL2:181
;NOZZLE_DIAMETER:0.4
;MINX:67.036
;MINY:100.029
;MINZ:0.3
;MAXX:110.768
;MAXY:126.971
;MAXZ:5.4
;Generated with Cura_SteamEngine 4.4.0
M82 ;absolute extrusion mode

G92 E0 ;; not necessary, but no problem
----- Enable Extruder 1 --------
;----------------------- no explicit selection of first extruder here. no big deal.
G92 E0
G10 ;--------------------------------------- Retract. no problem here, we are un-retracting soon!
;LAYER_COUNT:42
;LAYER:0
M107 ; disable fan
G0 F6000 X88.692 Y103.418 Z0.3 ;---------------- goto start position
;TYPE:SKIRT
G11 ;---------------------------------------------------- un-retract, then start printing
G1 F3000 X89.291 Y102.903 E0.09479
G1 X89.939 Y102.452 E0.18953
G1 X90.63 Y102.068 E0.2844
G1 X91.355 Y101.756 E0.37911
[... omitted lines ...]
G0 F6000 X95.238 Y115.603
G1 F3000 X96.395 Y116.76 E157.61981
G0 F6000 X95.829 Y116.76
G1 F3000 X95.238 Y116.169 E157.72011
G1 X95.168 Y116.099 ;----------------- end of slice for object 1
G10 ;---------------------- retract at first extruder. sensible, as we switch to second extruder soon
G1 F600 Z1.3 ;------ z-lift
G92 E0
; ====== Disable Extruder 1 ======
T1
G92 E0
; ====== Enable Extruder 2 ======

G10 ;------------------ retract. this one is a problem: the filament of this extruder already is retracted
;MESH:a.stl
G0 F6000 X77.168 Y116.099 Z1.3
G1 F600 Z0.3 ;------------ move to starting point
G0 F6000 X75.404 Y116.099
G0 X67.069 Y117.931
;TYPE:SKIRT
G11 ;----------- un-retract
G1 F3000 X67.069 Y114.13 E0.45612
G1 X67.036 Y113.737 E0.50345
G1 X67.045 Y112.889 E0.60521
G1 X67.069 Y112.659 E0.63296
G1 X67.069 Y109.069 E1.06376
[... omitted lines ...]
G0 F9000 X83.028 Y116.76
G1 F3000 X83.76 Y116.028 E48.67077
G0 F9000 X83.76 Y116.594
G1 F3000 X83.594 Y116.76 E48.68016 ; ------- end of slice for extruder 2
G10 ;----------------------- retract from extruder 2. fine, since we are about to change to extruder 1
G1 F600 Z1.4
G92 E0
; ====== Disable Extruder ======
T0
G92 E0
; ====== Enable Extruder ======

G10 ;------- retract from extruder 1. at this point, we have retracted this extruder twice
;MESH:b.stl
G0 F9000 X101.594 Y116.76 Z1.4
G1 F600 Z0.4
G0 F9000 X101.594 Y113.155
G0 X99.025 Y112.975
;TYPE:WALL-INNER
G11 ; ------------ un-retract, does cancel out last retract, but not the retract from switching to ext 2
G1 F2276.7 X99.025 Y109.669 E0.13224
G1 X102.331 Y109.669 E0.26448
G1 X102.331 Y112.975 E0.39672
[... omitted lines ...]
```

**Expected results**

On each extruder change, the last material should be retracted and the new material should be un-retracted.
```
;FLAVOR:UltiGCode
;TIME:427
;MATERIAL:294
;MATERIAL2:181
;NOZZLE_DIAMETER:0.4
;MINX:67.036
;MINY:100.029
;MINZ:0.3
;MAXX:110.768
;MAXY:126.971
;MAXZ:5.4
;Generated with Cura_SteamEngine 4.4.0
M82 ;absolute extrusion mode

G92 E0 ;; not necessary, but no problem
----- Enable Extruder 1 --------
;----------------------- no explicit selection of first extruder here. no big deal.
G92 E0
G10 ;--------------------------------------- Retract. no problem here, we are un-retracting soon!
;LAYER_COUNT:42
;LAYER:0
M107 ; disable fan
G0 F6000 X88.692 Y103.418 Z0.3 ;---------------- goto start position
;TYPE:SKIRT
G11 ;---------------------------------------------------- un-retract, then start printing
G1 F3000 X89.291 Y102.903 E0.09479
G1 X89.939 Y102.452 E0.18953
G1 X90.63 Y102.068 E0.2844
G1 X91.355 Y101.756 E0.37911
[... omitted lines ...]
G0 F6000 X95.238 Y115.603
G1 F3000 X96.395 Y116.76 E157.61981
G0 F6000 X95.829 Y116.76
G1 F3000 X95.238 Y116.169 E157.72011
G1 X95.168 Y116.099 ;----------------- end of slice for object 1
G10 ;---------------------- retract at first extruder. sensible, as we switch to second extruder soon
G1 F600 Z1.3 ;------ z-lift
G92 E0
; ====== Disable Extruder 1 ======
T1
G92 E0
; ====== Enable Extruder 2 ======

;MESH:a.stl
G0 F6000 X77.168 Y116.099 Z1.3
G1 F600 Z0.3 ;------------ move to starting point
G0 F6000 X75.404 Y116.099
G0 X67.069 Y117.931
;TYPE:SKIRT
G11 ;----------- un-retract
G1 F3000 X67.069 Y114.13 E0.45612
G1 X67.036 Y113.737 E0.50345
G1 X67.045 Y112.889 E0.60521
G1 X67.069 Y112.659 E0.63296
G1 X67.069 Y109.069 E1.06376
[... omitted lines ...]
G0 F9000 X83.028 Y116.76
G1 F3000 X83.76 Y116.028 E48.67077
G0 F9000 X83.76 Y116.594
G1 F3000 X83.594 Y116.76 E48.68016 ; ------- end of slice for extruder 2
G10 ;----------------------- retract from extruder 2. fine, since we are about to change to extruder 1
G1 F600 Z1.4
G92 E0
; ====== Disable Extruder ======
T0
G92 E0
; ====== Enable Extruder ======

;MESH:b.stl
G0 F9000 X101.594 Y116.76 Z1.4
G1 F600 Z0.4
G0 F9000 X101.594 Y113.155
G0 X99.025 Y112.975
;TYPE:WALL-INNER
G11 ; ------------ un-retract

G1 F2276.7 X99.025 Y109.669 E0.13224
G1 X102.331 Y109.669 E0.26448
G1 X102.331 Y112.975 E0.39672
[... omitted lines ...]
```

**Project file**
[test.zip](https://github.com/Ultimaker/Cura/files/4006475/test.zip)

StephanRichter, 2019-12-30:
Is there a possibility to flag this issue as bug, or is this to be done by the repo owener?

nallath, 2019-12-30:
You should also be able to flag it, but I've just done that for you.

nallath, 2019-12-30:
As for your issue, isn't it a firmware issue? It seems that the g10 command is not handled quite right. If it's already unretracted, so it should do nothing.

StephanRichter, 2019-12-30:
Maybe both. On the one hand, the firmware ( Marlin, the version shipped with the Cura 4.4.0 AppImage) should not retract, if the material already is retracted.

On the other hand, I think the number of un-retractions in a G-Code file *should* match the number of retractions, shouldn't it?

nallath, 2019-12-30:
I don't think any changes will be made to the UM2.1 firmware anymore (and it's even less likely if it's to support a modification).

StephanRichter, 2019-12-30:
Hm. Had a closer look into the firmware directory shipped with the latest AppImage. There is a file "Attribution.txt" stating that the image is build from https://github.com/Ultimaker/SecretMarlin which redirects to https://github.com/Ultimaker/Ultimaker2Marlin . I think I will try to build this firmware, and if I succeed, I will have a look into the code and see if I can mitigate the problem there.

Still, I think the G-Code generated by Cura should not contain more retractions than un-retractions.

Ghostkeeper, 2019-12-31:
Cura makes the assumption that after your start g-code, the material is at the tip of the nozzle (not retracted). However when a nozzle has finished printing it doesn't leave it unretracted, since that would ooze on the print. I'd expect there to be one more retraction than the number of unretractions, for each extruder.

However Cura works internally with E positions rather than `G10` and `G11`. There could still be something wrong with those, since Cura will track the E position, not the retracted/unretracted boolean state or anything. It's more complex than that.

StephanRichter, 2019-12-31:
> Cura makes the assumption that after your start g-code, the material is at the tip of the nozzle (not retracted). However when a nozzle has finished printing it doesn't leave it unretracted, since [...]

That is exactly what I would expect, too. However, the number of retractions is is larger than the number of unretractions by the number of layers. In other words: for each nozzle change, (one per layer) an additional retraction occurs:

Layers in test object: 98
```
srichter@twoflower:~/tmp$ cat test2.gcode | grep G11 | wc -l
932
srichter@twoflower:~/tmp$ cat test2.gcode | grep G10 | wc -l
1031
```

Ghostkeeper, 2020-01-02:
We've decided to defer this issue. The occurrence of this bug is really slim right now. If we get more users that have multi-extrusion printers with firmware retractions, we'll reconsider that priority. Of course if you find the problem in CuraEngine we can also take a look at merging a fix. We won't get to spend time on fixing this any time soon.

StephanRichter, 2020-01-02:
Unfortunate, but understandable. Anyways, thank you for your efforts!

closed 2023-06-24 by github-actions[bot]