Ultimaker/Cura #6864: [4.4.1] Cannot create or duplicate material profile when installed over version 4.4.0

Nuttavoot, 2019-12-28:
**Application version**
4.4.1

**Platform**
Windows 10

**Printer**
Creality Ender-5 or any printer

**Reproduction steps**
1. Manage materials.
2. Select Generic PETG
3. Click Duplicate.

**Actual results**
Nothing.

**Expected results**
A duplication of PETG profile.

**Additional information**
Also try clicking create, nothing happened. If do a clean install of version 4.4.1 everything works fine until I restore a backup from Cura Backup Extension then same problem as above again.

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4006902/cura.log)

Nuttavoot, 2019-12-28:
Found the cause: Uninstall Print setting plug-in solved the problem.

closed 2019-12-28 by Nuttavoot

fieldOfView, 2019-12-28:
As the author of both the Printer Settings plugin and the Material Settings plugin, which one did you (un)install?

fieldOfView, 2019-12-28:
See https://github.com/fieldOfView/Cura-MaterialSettingsPlugin/issues/7

Nuttavoot, 2019-12-28:
> As the author of both the Printer Settings plugin and the Material Settings plugin, which one did you (un)install?
> 
> Edit: looking at the log file, it seem you meant the Material Settings plugin.

Yes, it was Material Setting plugin.