Ultimaker/Cura #6866: Different speeds for outside wall "out" and outside wall "in" of object for reduced print time.

ossvr, 2019-12-28:
When printing objects such as a hollow cylinder (example), Cura applies the same speed to all exterior walls (see annex- "wall A" and "wall B"). My suggestion is to create an option to apply different speeds for these two types of wall, as wall "B" will not always be visible, not requiring the same print quality as wall "A", especially if the object receives something like a lid. Therefore, printing time may be significantly reduced.
[image]

Ghostkeeper, 2019-12-31:
What would you expect to happen if there is a tiny sliver missing from the cylinder in that case? Like a C shape instead of an O shape? Wall B becomes Wall A then. Are they both outside or both inside?

This feature can be done, but I feel like it's a bit specific to just your print and might not get much use.

Ghostkeeper, 2019-12-31:
You can by the way already do this by setting the Small Hole Max Size setting to be between the diameter of the hole and the diameter of the outer contour, and then adjusting the Small Feature Speed. However you can't adjust it above 100% which looks like a settings mistake to me. I'll remove that restriction.

ossvr, 2019-12-31:
Hi. My suggestion can be useful for any object whose interior need not be visible. There are many objects that are printed with this need. I do not believe it would be a very restricted option. As identification features, the walls "A" have concave angles and longer length than the walls "B", besides these have convex angles.

ossvr, 2019-12-31:
I showed the cylinder as a simple example. Holes and cuts in the wall would not prevent identification of walls "A" and "B".

Ghostkeeper, 2020-01-02:
I've discussed this with the Cura team and we've decided that this could be useful for productivity. It's been put on our backlog.

ossvr, 2020-01-03:
Thank you for your attention and wish you success for the team.

Ghostkeeper, 2021-06-08:
Our PO has reconsidered it and removed it from the planning. We're unlikely to get around to implementing this soon, so we'll mark it as deferred.

njsokalski, 2025-06-21:
I would find this (or something similar) to be very useful. The basic idea of my request would be to have all (or any appropriate) places that involve wall properties to reference them as "Concave Walls" or "Convex Walls". I think this would be very useful, especially in scenarios involving holes and enclosed areas.