Ultimaker/Cura #6867: Fails to launch cura 4.4

Vertabreak, 2019-12-28:
**Application version**
cura 4.4

**Platform**
ubuntustudio 19.10 
GTX 1070

**Printer**
none wont even open

**Reproduction steps**
installed 
attempted to launch

**Screenshot(s)**
nothing to show as nothing happens 

**Actual results**
click to launch = nothing

**Expected results**
cura open

**Project file**
none

**Log file**
/usr/lib/python3/dist-packages/apport/report.py:13: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import fnmatch, glob, traceback, errno, sys, atexit, locale, imp
Traceback (most recent call last):
  File "/usr/bin/cura.real", line 15, in <module>
    import sentry_sdk
ModuleNotFoundError: No module named 'sentry_sdk'

**Additional information**
none

smartavionics, 2019-12-28:
I don't think the sentry stuff has made it yet into an official Cura release yet so the Cura you are using must be based on the UM master branches. Whoever built your Cura needs to also update their build environment (see the cura-build-environment repo for details) to include the sentry sdk.

Vertabreak, 2019-12-28:
i just rolled back to 4.2 for now just trying to make a note of the issue.

nallath, 2019-12-30:
The cura-build-environment is already updated to have the sentry_sdk as a package. 4.4 does not have the sentry_sdk, so it's a bit strange that it's asking for it (as smartavionics already mentioned, this is not a 4.4 build but a master build)

closed 2019-12-30 by nallath