Ultimaker/Cura #6868: 'Custom selection' settings get overridden when 'Show All Settings' is used

vvranjek, 2019-12-29:
**Application version**
4.4.1

**Platform**
Ubuntu 18.04

**Printer**
Anycubic MEGA i3

**Reproduction steps**
1. Go to 'Manage Settings Visibility' and check any settings that you want to be visible under 'Custom Selection'

2. Now under print settings you can change the visibility using 'Basic', 'Advanced', ' Custom selection', etc... and everithing is fine until...

4. Chose 'Custom Selection'

5. Then chose 'Show All Settings'. 

**Actual results**
At this point all of the custom setting under 'Custom Selection' get overridden by showing all settings and you have to manually go through step 1.

**Expected results**
If I want to show all setting under 'Custom selection' I can go to 'Manage Settings Visibility' and set 'Check all' checkbox and that's fine. The way I described in step 6 is not fine.

Ghostkeeper, 2019-12-31:
We still have the wish to add visibility profiles at some point. Currently linked to this issue: https://github.com/Ultimaker/Cura/issues/5047

We have 3 profiles at the moment: Basic, Advanced and Expert. The option you're using does literally what it says on the tin. It's behaving exactly as if you're manually checking each setting's visibility. It's expected behaviour right now, so I'll change this to a feature request.

I think you should look into the Setting Visibility Set Creator plug-in on the Marketplace. It can do what you want to do and more.

nallath, 2020-01-02:
I've added the "all" setting visibility preset in 8b393317ff1ca5024b203f2fe85642a3bf5334f4

closed 2020-01-02 by nallath