Ultimaker/Cura #6869: ENDER 3 print setting profile wrong and damage the printer

Skybeat, 2019-12-29:
**Cura 4.4.1**
(The version of the application this issue occurs with.)

**Windows 10 x64**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Ender 3**
(Which printer was selected in Cura?)

**Reproduction steps**
1. Add a Ender 3 printer in Cura Setup
2. try to print

**Screenshot(s)**
actual Screenshot from Cura 4.4.1
[image]

Screenshot from Cura 4.1.0
[image]

**Actual results**
if i add Ender 3 in Cura 4.4.1 and try to print, the nozel touches the bed and the printer try to go down more.
I have installed a BLTouch.

With Cura 4.1.0 the Ender 3 printer settings worked very well and fine, with Cura 4.4.1 it is not posible to start a print and more, with Cura 4.4.1 and the Ender 3 profile, my printer will be damaged!!!!

**Expected results**
expected result will be that my Ender 3 works as fine as it does in Cura 4.1.0

**Project file**
n/a

**Log file**
n/a

**Additional information**
if i try to change the values in Printhead Settings for "x min", "y min" and "y max" the vuales reset all the time to the initial values from the profile, so I'm not able to change this settings. Even more,  a smal image what shows the directions for this values will be very nice since it is changed to polygone!

So, how is it posible to setup the Ender 3 in Cura 4.4.1 and bring it to work whiout any damage on the printer himself?!

Ghostkeeper, 2019-12-31:
If you think the defaults in Cura are wrong and you've tested better ones, we'd love to hear about those. If you make a pull request we'll review your proposed changes.

In between the lines I'm reading that you think the print head shape is wrong (and indeed that could damage the printer if it is). You're reporting a bug that you can't change them. I'm not able to reproduce that bug. I've went into the menu and changed the print head settings, and it seems to change fine:

[image]

Your reproduce steps aren't reproducing the issue here, at least not without the additional step of "Buy an Ender 3 and wait some weeks for it to ship from China." How can I see what Cura is doing wrong without needing to buy all of the 180 printers that Cura has profiles for?

PeteBurgess, 2020-01-01:
I can reproduce this bug. Change the Y min to 20, move to the next box, Y min resets to whatever it was previously. It will let you change the X and Y min to a negative integer or a 0, but not a positive integer.

Ghostkeeper, 2020-01-01:
I also tried to reproduce it on Windows with the 4.4.1 build but the print head size can be modified fine for me.

Maybe it depends on the localisation? What locale are you using in Windows?

Traste, 2020-01-28:
+1
I just upgraded from Cura 4.4.0 to Cura 4.4.1.

The reason for the settings not being possible to edit (on Windows machines) appears to be related to permissions - if i launch Cura as an Administrator, I can edit them without issue.

The actual measures given appears to correspond to earlier settings and correct, so if someone can concur on that, it would be nice.

NCLMChicago, 2020-04-19:
I'm having the same problem on with both the CR-10 and Ender-3 profiles.  Running as administrator did not help.  The same problem still exists in the 4.6 beta.  They both share the same printer profiles so I am wondering if something as simple as storing those files in a different location would help.

Ghostkeeper, 2020-04-22:
Sorry, I meant to say "locale", not "localisation" or "location". It's like language, but also including the conventions to write date, time, decimal numbers, etc. It's not about where the files are.

NCLMChicago, 2020-04-25:
Here are my settings . . . 
[image]

Ghostkeeper, 2020-06-29:
That's the same locale as what I was using on Windows, too. However I'm not able to reproduce the problem.

SkyeShark, 2020-08-18:
I have this same bug.

NCLMChicago, 2020-08-18:
Its really strange this keeps getting closed even with so many of us
reporting the problem.  The Ender 3 is a fairly popular printer.

nallath, 2020-08-18:
Keeps getting closed? It was closed exactly once when we introduced the no-response bot.

The ender is indeed a popular printer, but since all development is currently paid for by Ultimaker, changes / fixes for those machines simply get preference. 

We've also not really been able to reliably reproduce this on our end, which also makes this rather tricky to fix (since we wouldn't even know if our changes fix the issue)

NCLMChicago, 2020-08-18:
No disrespect intended.  I appreciate all the hard work being done.  If
there is anything I can do to help, please let me know.

DaveP-Hawaii, 2020-08-18:
I am brand new to the 3D Printing world and had downloaded the Cura slicer program 4.6.2. I found this forum when trying to find printer settings for the Ender-3 Pro. I, too, have negative numbers X and Y min and cannot change them to positive numbers. After reading the first post, I am afraid to print my first item for fear of damaging the printer.

I also tried to add a Custom FFF Printer to input the data and cannot change the X and Y min to positive numbers there either.

Ghostkeeper, 2020-08-19:
Well, we'd need to know what the correct values would be for these settings, so that it forces the user to keep sufficient distance between printed objects.

DaveP-Hawaii, 2020-08-19:
Sorry, I'm new to 3D printing and jumped into this discussion too soon. I
didn't understand what the "Printhead Settings"  in Cura ment, but after
doing some more investigating I see that they are for managing horizontal
spacing from the nozzle housing to surrounding objects when printing more
than one object, one at a time. As far as I'm concerned, these settings are
correct and only need to be changed if modifying the nozzle housing.

Thank you for your response.

Ghostkeeper, 2020-08-21:
All right, thanks for the help then :)

Ghostkeeper, 2020-08-24:
> Unless Cura changed the way the print head settings are calculated in
different versions, the numbers I gave remain the correct ones for the
Ender 3.

> Must the X min and Y min be negative?

In fact this is something that was changed recently-ish. Could have been 4.4 in fact; I don't remember when exactly. But yes, the X min and Y min must be negative now. They represent the position of the left and front sides of the print head, where 0,0 is the print head position (normally the same as the nozzle position with single-extruder printers). When we changed that, we should also have changed the values for all printers that existed at the time.

NCLMChicago, 2020-08-25:
So should DavidP and I both test the defaults as-is?  Perhaps his is just
confusion in the discussion groups because people didn't notice the change
in calculating the x min and y min?  Could the help text that pops up
explain that they "must be negative"?  (I don't have my machine in front of
me to check the actual help text.)

Ghostkeeper, 2020-08-27:
The text box won't allow you to type a positive number there, so it should be pretty clear I think.

drdurkee, 2021-05-12:
I'm interested in working on documentation for this change post v4.4 where it now only possible to input min values as negative numbers.   My rough understanding is that we assume the nozzle tip is 0,0,0 in the plane and so measuring a min, for x and y put them both as negative number. 

... found this thread while researching how to get positive settings in for the XYZ DaVinci Jr.   Since it is not possible to get positive numbers, its clear that something changed since June 20, 2020 when XYZ set up there Printing with Cura page: https://www.xyzprinting.com/en-US/news/printing-with-cura 

(it would be helpful to get some clarification and guidance about what needs to be done to make it work instead of a bug that needs to be fixed)  

... according to the XYZ direction, one creates a custom FFF profile, and then populates it with printhead setting from: https://xyzofficialweb.blob.core.windows.net/xyz-v3/download/Cura_Profile_XYZ.PDF  -- all of the directions for their nano, mini and jr still reference X min 10mm and Y min 10mm.  

... would really like to be able to reach out XYZ and point to some reference blog, wiki, or even this discussion thread to show how the new settings should be calculated, as they say to use the latest version of Cura, yet this is not possible post v4.4.  I did measure the distances following the procedure sketched out by DaveP-Hawaii: 

X min -24
Y min -40
X max 45
y max 24

I'm running... 
Winx64 Cura v4.9.1 (DaVinci Jr, 1.0) - yes, I know it's not Ender3, but this seems to be a case where there needs to be some documentation which I am happy to help work on.

Ghostkeeper, 2021-05-19:
I'd say this could fall within the scope of the Settings Guide. It's not currently there because these 4 numbers are not actual "settings". They are fields that are used as a more convenient way to configure the [Head & Fans Polygon](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/machine_settings/machine_head_with_fans_polygon.md). Internally the setting can be any polygon but it's hard for a user to write down a list of arbitrary coordinates. The 4 numbers in the Machine Settings dialog define a rectangular polygon, but it still works the same way. The coordinates of the rectangle are still relative to the position of the print head (which is usually the same as the position of the first nozzle).

NCLMChicago, 2021-05-19:
Thanks so much for your time and attention to this.  At least some of us
found the current state of affairs confusing and I believe this would help.

krikru, 2021-06-03:
Why do the default settings in Cura 4.9.1 say that X (Width) and Y (Depth) are both 235.0 mm? The print size is 220x220x250 mm (https://www.creality3dofficial.com/products/official-creality-ender-3-3d-printer), so shouldn't it say 220.0 instead of 235.0?

And now when I have added a printer profile, how can I remove it again? The choice "Remove" is greyed out so it seems I cannot remove it. The fact that the X and Y settings seem incorrect out of the box is slightly unnerving, because if these are incorrect, other settings may very well be incorrect too, and I don't want to have a printer profile that is likely to contain incorrect settings that may even damage my printer if I try to use it.

Ghostkeeper, 2021-06-14:
> Why do the default settings for Creality Ender-3 in Cura 4.9.1 say that X (Width) and Y (Depth) are both 235.0 mm? The print size is 220x220x250 mm (https://www.creality3dofficial.com/products/official-creality-ender-3-3d-printer), so shouldn't it say 220.0 instead of 235.0?

The reason for that, as best as we know it, is explained in this commit message: https://github.com/Ultimaker/Cura/commit/726561bf309616107ad874d4e508b3a3ec9e5701 and heavily discussed here: https://github.com/Ultimaker/Cura/issues/6267 . There is a lot of discussion and theory behind this.

> And now when I have added a printer profile, how can I remove it again? The choice "Remove" is greyed out so it seems I cannot remove it.

You can't delete the printer that is currently active. You'll need to activate a different printer first.

Ghostkeeper, 2021-06-14:
We are not able to change the defaults, if they are necessary, so to fix this ticket we'll need more information on what the correct settings would be to not damage the printer.

krikru, 2021-06-14:
> The reason for that, as best as we know it, is explained in this commit message: [726561b](https://github.com/Ultimaker/Cura/commit/726561bf309616107ad874d4e508b3a3ec9e5701) and heavily discussed here: #6267 . There is a lot of discussion and theory behind this.

Thanks for the explanation. Perhaps it would be good to provide some explanation for this in the program too, to avoid confused users like me? How can that be done?

Ghostkeeper, 2021-06-22:
We don't really have a place for printer-specific explanations at the moment.

Does it need an explanation? The size should be what the stock firmware allows. If the firmware does not agree with the Creality website, I think that is a problem on their side that is a bit out of scope for Cura to rectify.

closed 2021-06-28 by no-response[bot]