Ultimaker/Cura #6870: How to create an all-in-one file with printer definition?

MarkMan0, 2019-12-29:
**Application version**
4.4 and above

**Platform**
Any

**Printer**
**Reproduction steps**
**Screenshot(s)**
**Actual results**
**Expected results**
**Project file**
**Log file**
N/A

**Additional information**
Hello! 
I work for a company, and we would like to manufacture 3D printers. How can I create an importable file, which contains all the definitions of the printer and one or more profiles? 
I've looked at the _resources/definitions_ folder, and I an create a json for our printer.

Is it possible, to create a file, that the user would just drag&drop into cura, and he could than just select/import the correct printer and be done?

nallath, 2019-12-30:
Nope, this isn't possible. The only way to get new machines into Cura is currently by pull request.

closed 2019-12-30 by nallath

fieldOfView, 2019-12-30:
Can't the .curapackage format be used to install definitions and quality profiles to the configuration folder?

nallath, 2019-12-30:
We might be able to use this, but then we have some issues with the fdm_materials. Let's say I want to add the amazingNallathPrinter™. I can easily create a package that has the extruder, definition and a bunch of quality files. But if I also want some settings to be different for generic_pla, that's suddenly an issue. I can't just override the existing material (because that would fail if someone else does the exact same thing). 

Our initial idea was to add some sort of material extension so that materials can be appended one way or another, but the xml material profile is already a dragon as is, so doing that right will take quite a bit of work.