Ultimaker/Cura #6871: Need support for new printer with old version (32bit) Cura

N1MKG, 2019-12-29:
I've downloaded the final version that can run under 32bit (I'm using Win 7 32bit exterprise.)
It does not list my printer Monoprice MP10 (Same as the Malyan Ma10 printer), yet the
latest Cura version does list this printer.

Is there any printer (there are many) listed in the old 32bit software that are compatible
with this new printer?

Please also email to "grossman19@gmail.com"
Thanx

rburema, 2019-12-31:
You can start with the 'Custom FFF' printer, but you need to fill in the values for your printer then.

Ghostkeeper, 2019-12-31:
We don't have plans to update the 32-bit version of Cura any more. It was a lot of work to maintain that and didn't have enough users, so I'll close this ticket as a won't-fix.

There is no definition in Cura 2.3 that is tested to be compatible with your printer. It did have a Malyan M180 iirc, but your best bet is probably to use the Custom FDM Printer (as it was called then).

closed 2019-12-31 by Ghostkeeper