Ultimaker/Cura #6872: Failed to start AppImage on some debian buster machines

magithub1, 2019-12-29:
Cura-Version 4.4.1 and 4.4.0

System-version
linux, debian buster 64bit, AppImage

Start with 
./Ultimaker_Cura-4.4.1.AppImage
and after some seconds the process is crashed.
The log is seen in
[stderr.log](https://github.com/Ultimaker/Cura/files/4008447/stderr.log)

Then I extract the AppImage and start with
strace -f -o strace.log ./AppRun
The result is seen in
[strace.log](https://github.com/Ultimaker/Cura/files/4008454/strace.log)

On two machines I got this error and on two machines the AppImage is running. I can't find out the difference between the installations. AppImages should start on all machines and test with a other appimage show that all my machines can start appimages. Only with cura I got the problem.

What can I do?
with regards

smartavionics, 2019-12-29:
Hi, looking at your strace.log it seems to me that you have a glob.py file in your home directory which it is loading instead of the file in the AppImage?

````
12456 stat("/home/andreas/glob.py", {st_mode=S_IFREG|0644, st_size=1230, ...}) = 0
12456 stat("/home/andreas/glob.py", {st_mode=S_IFREG|0644, st_size=1230, ...}) = 0
12456 openat(AT_FDCWD, "/home/andreas/__pycache__/glob.cpython-35.pyc", O_RDONLY|O_CLOEXEC) = 7
````
If that's the case, then it's not good that the AppImage loads python files from outside of the AppImage.

magithub1, 2019-12-30:
Hallo Mark,

you are great! After deleting |/home/andreas/glob.py and
||/home/andreas/__pycache__/glob.cpython-35.pyc| I must once start
./AppRun in extracted files and after that every time
./Ultimaker_cura_4.4.1.AppImage start without problems too. For me is
that solved, but I think the AppImage must be reengineered.

With regards

smartavionics, 2019-12-30:
Yes, keep this issue open so that the Cura devs see it and they can think about stopping Cura from loading Python modules from outside of the cura AppImage.

Ghostkeeper, 2019-12-31:
Yeah Python is a bit complex in the places it will try to import from. There's a tutorial here: https://github.com/AppImage/AppImageKit/wiki/Bundling-Python-apps

It's something we'll need to improve I think.

LipuFei, 2020-01-02:
For dev, JIRA ticket: CURA-7081

evtrados, 2020-03-03:
This issue has been resolved and the fix will be in the next stable release.

closed 2020-03-03 by evtrados