Ultimaker/Cura #6873: Crash on Custom Filament Form

roberto081194, 2019-12-29:
**Application version**
4.3.0

**Platform**
Windows 10 version 1809 Build 17763.914

**Printer**
Ender 3

**Reproduction steps**
1. Go to preferences -> Material (Ctrl+K)
2. Create new custom material.
3. Type on brand 'Generic'
4. Go to next field

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
It crashes unexpectly.

**Expected results**
Can type or continue making actions in order to save the form.

**Log file**
[error.log](https://github.com/Ultimaker/Cura/files/4008471/error.log)
 to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

nallath, 2019-12-30:
Does this issue also happen with 4.4.1?

roberto081194, 2019-12-30:
Yes, it happens too.

nallath, 2019-12-31:
I don't see anything in the logs about a crash and I can't reproduce it myself (although I am on Fedora, not Windows)

closed 2020-06-19 by no-response[bot]