Ultimaker/Cura #6874: Extreme Priming After Start G-Code

bmxorcist, 2019-12-29:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

nallath, 2019-12-30:
Nope. The only thing I can come up with is that it's not liking the negative extrusion, but that's more of a printer issue than one with Cura.

Ghostkeeper, 2019-12-31:
Underflow? I think you're looking at a firmware bug here.

bmxorcist, 2019-12-31:
I would think that too, but the same firmware works flawlessly on any other slicer. Anyway, I've moved to Prusa Slicer and I don't think I'll ever turn back. This program is great.

Ghostkeeper, 2019-12-31:
So if you let another slicer write the same g-code to a file and send that over to the printer it works?

There must be something different in the g-code.

bmxorcist, 2019-12-31:
I agree. I think it's something Cura does before or after sending the starting gcode. Could it be the M82?
But I've reflashed my firmware and it's running beautifully with Prusa Slicer, and I do still have the gcode in question... I'll run it and see if it still spews on the new firmware and running off the SD instead of over USB.

Ghostkeeper, 2020-01-01:
It could be the M82 since that should change how extrude commands are interpreted. If it is though that is still a firmware bug. Since Cura sets the current E position to 0 at the beginning of the print with the `G92 E0` commands, it shouldn't matter whether it's relative or absolute. `0 + -1` is still `-1` after all.

FlailAway, 2020-01-02:
Not sure if it is any help, but that E-1 works fine as a blob-suck at the start on a stock A6 board and a Delta with Gen-L both running Marlin 1.1.9. I actually use it in my end-code to release pressure on the filament in Bowden on both. Been using it for a looong time.

PrusaSlicer? You will be back. :)

closed 2024-12-07 by github-actions[bot]