Ultimaker/Cura #6879: Print Sequence Optimisation

siteswapjuggler, 2019-12-29:
**Is your feature request related to a problem? Please describe.**

1. When "one at a time" is selected the slicer is unable to slice when the clearing areas around parts are going outside the bed.

2. When "one at a time" is selected the slice doesn't want to slice when objects are higher than the gantry height.

**Describe the solution you'd like**

1. "one at a time" should not take care of clearing area outside the bed but only intersection with other clearing areas

2. a smart feature would be to allow high part one at a time by going front to back (or back to front) for machines with a moving buildplate. 

3. different machines should have different "one at a time strategies" : Core XY, Core Y, Delta

**Affected users and/or printers**

Advance user mostly, but potentially every one and every printer is affected.

nallath, 2019-12-30:
>     1. When "one at a time" is selected the slicer is unable to slice when the clearing areas around parts are going outside the bed.

This has been fixed for 4.5

>     3. different machines should have different "one at a time strategies" : Core XY, Core Y, Delta

Since Ultimaker printers all have the same gantry setup, It's not likely that we will put time into changing it so that this behavior can be dynamic (especially since it's not a trivial change to do so). But as usual, just because we probably won't put any time in it, it doesn't mean that pull requests aren't welcome.

siteswapjuggler, 2019-12-31:
I understand, shouldn't it still be an Idea to extend the one at a time strategy allow the print of two parts in opposite corners. If I'm not mistaken it should be no collision problem for CoreXY configuration and would be a direct benefit for Ultimaker users.

Ghostkeeper, 2019-12-31:
It's hard to guarantee that it won't collide then, for instance when making a diagonal movement, moving the build plate down while homing the print head.

siteswapjuggler, 2019-12-31:
hmm apart from the end gcode, what could possibly go wrong ? if we start bottom left, then up right we also let the extruder swap available on S5... we then just need to post process the endgcode avoid movement that would collide the parts... Honnestly think it could work.

Ghostkeeper, 2020-01-01:
I agree that could work for most printers. However Cura will currently adjust the print order to maximise the available build plate space, so it will increase the collision area of the objects if the order were to be fixed, such as from front left to back right.

siteswapjuggler, 2020-01-02:
Let's keep that somewhere, this is not a must have feature just a proof of concept that could work. Thanks for your time answering this.

closed 2023-07-13 by github-actions[bot]