Ultimaker/Cura #6880: A Palette Option

siteswapjuggler, 2019-12-30:
**Is your feature request related to a problem? Please describe.**

The problem actually comes more from Mosaic Manufacturing than Cura. It seems that newer version of Cura add some extra gcode that broke the Palette post-processing pipeline. (cf. https://support.mosaicmfg.com/hc/en-us/articles/115003212274?page=1#comment_360003748313)

**Describe the solution you'd like**

solution 1 : a post processing script or an experimental option that specify cura to slice with specification needed for palette post processing

solution 2 : some options to make multi color slicing easier in Cura, like an option to colorise part by extruder (no need to specify different material), one option to transform any possible machine into a palette compatible workflow (no need to add dummy extruder, just specify how many colors and color changing < this option can also be implemented for non palette user with a Change Filament at heigth automatically included wherever it's needed)

solution 3 : make cura to generate both gcode and palette splice data

**Describe alternatives you've considered**

Alternative 1 : Asking Mosaic to do a better intergration of Cura in their workflow, to share to specification of theire splice data.

Alternative 2 : taking time to implement a script or a post processing workflow by myself.

**Affected users and/or printers**

Multicolor improvement would be nice for every Cura user and every printer. Palette specific addition would mainly affect Palette user but would potentially make Cura a first choice for them.

Ghostkeeper, 2019-12-31:
> solution 2 : some options to make multi color slicing easier in Cura, like an option to colorise part by extruder (no need to specify different material),

Our current workflow allows the user to choose from several available colours via the material menu. I think it sort of makes sense that way. It's intuitive: The filament colour that you put on your printer can be selected from the menu where you choose your filaments. The problem you're facing, I think, is that many material manufacturers didn't bother to add materials for all of the colours in their arsenal. And so you can't choose because Cura doesn't know the colours available. As far as I know it's only Filo3D and Ultimaker materials that have multiple colours in Cura.

All of the Palette integration features you're proposing need access to a Palette to implement or even understand how it feels to use. I don't have such a modification so I can't do that for you, and neither can any of the other Cura developers. I think this is more appropriate for a plug-in, maybe. We won't implement this for you though.

closed 2019-12-31 by Ghostkeeper

siteswapjuggler, 2019-12-31:
It feels quiet a pain to work with actually but the possibility are endless. I'll see if I could work on some script or plugin to make a better workflow with cura and this device :) Thanks for you time and kind answers.

Ghostkeeper, 2020-01-01:
We've tried our best to make Cura extensible. Among other reasons, because there are so many types of printers and mods out there that it would be impossible for us to support all of it.

siteswapjuggler, 2020-01-02:
Make sure we owe you eternal respect for this. This is the main reason why I like Cura the most among all 3D Slicer out there. It's simple, open sourced and open to users suggestion and modifications via mods and plugins. 

I've worked on it yesterday and have now a clear picture of what will be my first roadmap on this, will post the result as soon as I got time to work on it and test it.

Have a nice day