Ultimaker/Cura #6881: post processing options improvements

siteswapjuggler, 2019-12-30:
**Is your feature request related to a problem? Please describe.**

PP Script are excellent but it's a pain to reconfigure them each time. And it's also a problem when we forgot to delete them from 1 print to another.

**Describe the solution you'd like**

Some solution that could work : 

1. post processing script are saved with cura project (and reload when open)
2. possibility to save pp presets and recall them
3. possibility to group pp scripts in the pp plugin
4. possibility to disable/enable script (or group of scripts) without deleting them

**Describe alternatives you've considered**

1. rewrite my own pp with my default parameters and cascading options

**Affected users and/or printers**

Everyone that use cura from middle users to expert.

Ghostkeeper, 2019-12-31:
> post processing script are saved with cura project (and reload when open)

This should work! They are saved with your printer, which is restored when you open a project file. Is this not working for you?

siteswapjuggler, 2019-12-31:
They are saved with the printer but not with the project. I made a video to try a better explanation.
https://www.dropbox.com/s/037b87rnccqgqes/PP_improvements.zip?dl=0

Ghostkeeper, 2020-01-01:
Funny how you can hear the fireworks going on in the background there!

As a workaround for "restoring" a (set of) post-processing scripts, what I often do is to create multiple printers and swap back and forth to enable or disable the post-processing scripts. We've chosen to connect the post-processing scripts to the printer because many post-processing scripts are made to fix compatibility with a certain printer. If you don't need those, then the other scripts are the ones made to adjust a certain print, and those usually don't need to be stored for long.

From your video I think I understood correctly what your bug report is, but indeed it should work like that: loading the project file should restore the printer including the post-processing scripts active on it. I can't properly test that right now but we'll try reproducing the bug at work.

siteswapjuggler, 2020-01-02:
That's actually why I tagged it as new feature more than a bug. I understand the origin of this when post-processing was mean to correct compatibility which explain why they are link to the printers, but also feel this isn't the only usage anymore. In fact we've got two usage : 

1. correcting gcode for specific machines (lets call it PPScripts group 1) (typically my palette script will go in this case)
2. adding specific action in specific project like temp towers, pause at height for nuts inclusion, filament change or other specific actions (lets call it PPS group 2)

PPS G1 is mean to be linked to the printer and it would be clearer to access, edit and see them in the machine settings window as a new tab of it like printer, extruder, pps.

PPS G2 is mean to be linked to projects as slicing paremeters are so perhaps we could think of something new to make a better display and workflow for them. What do you think of it ?

Ghostkeeper, 2020-01-03:
Displaying the post-processing scripts in the printer settings is an interesting idea.

For the second case we could attach the post-processing scripts to custom profiles, but that has two disadvantages:
* It only works when there is a custom profile.
* We can only load one set of post-processing scripts at a time, so it's either linked to the printer or linked to the custom profile, not both. We chose the printer originally because post-processing scripts that are print-specific don't usually need to be saved for long and can be discarded after the print is successfully finished.

Ghostkeeper, 2020-01-03:
For me the project file also restores the post-processing scripts. Here's a screen recording of this happening:

[project_file_postprocessing.zip](https://github.com/Ultimaker/Cura/files/4018750/project_file_postprocessing.zip)

siteswapjuggler, 2020-01-03:
Interessting (and nice tiger by the way ;)

 I did some more tests bases on the steps you made in the video it appears that post processing scripts are restored when a new printer is created when opening the project. It happens in your video because your first deleted the printer profile and I add the same behaviour here on my laptop. But it doesn't happened if I keep the printer profile and when I choose "update existing profile". Perhaps a clue.

Ghostkeeper, 2020-01-03:
That's it. If the printer is overridden it seems that it doesn't take the metadata changes along. It's a buggy-buggy-bug.

nallath, 2020-01-07:
This was a bit of a tricky issue, but it should be solved for 4.5. Thanks for reporting it!

closed 2020-01-07 by nallath

sunlanterns, 2020-01-25:
I've been running into this myself. Out of curiosity for when 4.5 is released should the project files I saved with pp scripts start pulling them in correctly or will I need to remake and resave?

Ghostkeeper, 2020-01-27:
Post-processing scripts that were saved in the project should be restored properly in 4.5. It was saving the post-processing scripts correctly in the files in 4.4, just not restoring them properly. So I expect that you won't need to resave your projects.

commit 523b4e96 (2020-01-07): Ensure that post processing scripts are correctly reloaded on project loading

Previously it would not re load post processing plugins if it was updating an existing machine

Fixes #6881