Ultimaker/Cura #6882: add brim in per modle settings

dsudweeks56, 2019-12-30:
**Is your feature request related to a problem? Please describe.**
I would like to add a brim to one part out of 13 parts on my build plate because it is very tall with a small adhesion base touching the build plate.

**Describe the solution you'd like**
in the Per Model settings box add an adhesion type pull down that would allow me to use 'brim' or 'none' (or maybe all the standard options for build plate adhesion,) for each selected part.

**Describe alternatives you've considered**
I usually build that one part separately with brim adhesion turned on. but I would rather build the complete set in the same build.

**Affected users and/or printers**
I am running an Creality Ender 3 Pro and a CR-10s Pro

Thanks for your consideration
Dave Sudweeks

nallath, 2019-12-30:
See https://github.com/Ultimaker/Cura/issues/1503 for the previous discussions regarding this feature.

Ghostkeeper, 2020-01-02:
We've decided to defer this again. It's not necessarily that we don't see the importance of this (would be nice to do) but the technical implementation of it is rather complex, so the effort to value ratio is still high.

The difficulty in implementing brim width per object is that brim lines currently merge when the objects are close together, and they'd have to start and stop rather than being closed contours if one object has a different number of brim lines than another.

disconn3ct, 2020-01-25:
FWIW looking at both discussions use cases, as well as my own, it seems like the simple option would solve many of the cases. Add a brim toggle to the per-model settings, using the overall settings for lines, distance, etc. (No customization, just on/off. Or reverse it with a toggle to block brim on a given model.) 

That resolves the number of lines issues, and doesn't account for (or allow) raft or other complex changes. It seems like it would solve the majority use case of "2 of my 8 objects need brims". (Anecdotally it seems like multipart designs are almost always a combination of huge sheets and pinpoint towers.)

I'm not sure how the internal geometry is managed, so this bit is probably worthless, but geometrically it seems reasonable to treat the override brims as part of the model shadow and not permit any funny business with merging lines or other efficiency goodness. It is ugly, but given the choice between ugly and missing I'll take ugly.

woschtkopp, 2020-04-01:
I would love to get this

daT4v1s, 2020-08-29:
another use cause would be to increase print area, in multi-part prints
this is because the brim reduces it (requires a margin to fit)
and removing brim from model that don't need it as much
may improve productivity

PhilBaz, 2021-11-12:
Its not that I want brim lines per object. just on or off per object. This would be fantastic where some parts are small and need brim, without wasting the brim for larger objects that don't need it. In my mind this is probably the main feature missing from cura.

Ghostkeeper, 2021-11-19:
Turning it on or off per object is algorithmically just as difficult as changing the number of brim lines per object. The [reason for the deferring](https://github.com/Ultimaker/Cura/issues/6882#issuecomment-570174003) still stands.

robinbanks420, 2022-04-28:
+1 would like this feature as well.

disconn3ct, 2022-04-28:
For anyone else who needs a solution, Prusa Slicer (and presumably related apps) can do it with no issues. Great for printing large pieces and assembly pegs at the same time.

PhilBaz, 2022-04-28:
@disconn3ct 
Ahh, thanks...Ill have to try it.

Much appreciated.

GrahamISW, 2022-05-18:
I love the tree supports in cura so like to use it for any of my slicing that needs supports, but I like the fact that in prusa/superslicer I can turn on and off settings per object on the plate.  For instance I am printing an axel that needs support  and a brim as it only has a small area for plate adhesion and overhangs that need support, but the gear I am printing at the same time is side and flat and doesn't need a brim or supports.  I'm still going to use Cura for the print, but its a pain in the neck to remove the brim from all the gear teeth. I could print each separately, but its inconvenient to have to set up and print twice.

BresCAD, 2022-07-29:
> For anyone else who needs a solution, Prusa Slicer (and presumably related apps) can do it with no issues. Great for printing large pieces and assembly pegs at the same time.

Hi, I very much agree with the request in this post. I tried using Prusa as you suggested, but I didn't find this feature at all, in fact I came across a very similar post to this one in the Prusa forums, asking for pretty much the same feature https://forum.prusa3d.com/forum/prusaslicer/brim-only-for-some-objects-on-the-build-plate/. 
Is there actually a way to do this? I would appreciate it a lot if you could share how you worked it out.
Thank you in advance.

disconn3ct, 2022-07-31:
https://help.prusa3d.com/article/skirt-and-brim_133969#brim-for-individual-objects The docs are better references than 3 year old forum posts :)