Ultimaker/Cura #6885: Unable to paste in new Start Code

grumpytwo, 2019-12-30:
**Application version**
Cura 4.4.1

**Platform**
Windows 10

**Printer**
Ender 3

**Reproduction steps**
1. Right clicked mouse 
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
nothing happens

**Expected results**
 I was expecting the usual cut paste copy menue

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

grumpytwo, 2019-12-30:
I have fitted a bed leveller to my Ender 3 and I am trying to paste the new code into the Cura start code only Cura won't let me.

I have copied the code from Notepad but there seems no way to paste it into Cure, right clicking the mouse and nothing happens.
I realise it is probably something I am doing wrong but I can't find an answer anywhere.

yet-another-average-joe, 2019-12-30:
just do Ctrl V...

Ghostkeeper, 2019-12-31:
Duplicate of https://github.com/Ultimaker/Cura/issues/6159.

closed 2019-12-31 by Ghostkeeper