Ultimaker/Cura #6887: "Full screen" window when open Cura

GooodFr, 2019-12-30:
Hello :)

I like so much Cura, and thank for free share !

When we open Cura, the windows seem "full screen" but is "reduced window".
So boring to reduce this windows : 

1. Reflex is to flip at in "reduced window", but that flip on "full screen"
2. Flip again to flip in "reduced window"
3. Still can't change the size, so move the window to a wrong place
4. Resize smaller to have a window able to fit in the screen
5. Move again to the correct place the window
6. Finally resize the window to the desired size

So PLEASE, could you change it ?

If you want that Cura open in full screen, the window could be in "full screen" and not in "reduced resized to full screen"
And the reduced size also could be saved

Regards,

Ghostkeeper, 2019-12-31:
Which operating system are we talking about here? Cura is expected to have a normal, resizeable and maximizeable window just luke any other application, and restore the position, size and maximize state when you restart it.

GooodFr, 2019-12-31:
So sorry, I'm on W10 64b
I have at least 2 screens

No, when I restart it, it's on a reduced window, but resized to fit entire screen.
Since always I think ...

fieldOfView, 2019-12-31:
On the "General" pane of the preferences, there's an option "Restore window position on start" that is checked by default. Could you uncheck it and then restart Cura? This should stop Cura from taking over your entire screen. If it does, it should be possible to resize and position Cura to taste, and then check that same option again. Cura should then open at your preferred location/size.

GooodFr, 2019-12-31:
Good !
I see "Restore window position on start", selected but no work.
All other apps work fine except Fusion 360, and Steam if there is an update, on 3 computers (Home, at work, my son computer)

"From Microsoft, App Windows that retrain their last position and size are programmed to do so. It is not a function of Windows."
https://answers.microsoft.com/en-us/windows/forum/windows_10-start-win_desk/restore-window-size-and-location/00df6b1b-289a-4c74-b4ad-585dc45ce604
And it seem that come from "software developers" …

Found !
By this I tried "Ctrl and close the window" and that work ! (also work for Fusion 360)
What do "Ctrl and close the window" ?
Now, when I exit Cura, Cura save the window position and size !

It seems that not a issue request, but a little bug.

Thank you for help !

closed 2019-12-31 by GooodFr

GooodFr, 2020-01-02:
No ... sometime Cura lost position and size ...
Today I open Cura, and it still open in "reduced window" resized in full sceen ...

"Restore window position on start" definitely don't work.