Ultimaker/Cura #6889: Cubic Subdivision Shell Options for Wall, Top, and Bottom Boundaries

mallstar2, 2019-12-31:
**Proposed Feature**
Include options to turn _Cubic Subdivision Shell_ on/off for wall, top, and bottom boundaries.

**Reason for Request**
I want to use Cubic Subdivision infill to save time and material on models that need high infill percentage only for top layer quality, but these savings are negated by the extra material that Cubic Subdivision adds near wall and bottom boundaries.

Ghostkeeper, 2019-12-31:
This is what Gradual Infill was made for. Did you try that? Does that not work for you?

mallstar2, 2019-12-31:
I have tried Gradual Infill but I had the same problem that was reported in issue #6001 
Cubic Subdivision doesn't have that problem because every infill line is inherently supported.

Ghostkeeper, 2020-01-16:
We've discussed this with the team, and while the feature is an improvement, we won't give it any priority now. Reasons:
* Gradual infill generally gives enough layers to bridge properly, given enough step height.
* Gradual infill will be faster to print.
* Gradual infill will work better with top surfaces with a lot of detail, since it doesn't work via cube subdivision.

This alternative isn't so bad, really. Try it!

It's not to say that this will never get built, but it's not something we'll look at now.

closed 2023-07-13 by github-actions[bot]