Ultimaker/Cura #6890: Minor Adjustments

BrittanGolden, 2019-12-31:
Numbering the Build plate for visual ref to makers trying to model items to scale or scaling for individual body types would be a serious step up for my business. 
Further more when replicating support void boxes they become real models and stick to the build plate. I like to take advantage of your angle degree scale to change the support dynamic but Im slowed down by the scaling process. 
Lastly and extremely important. When an object is scaled the change in its rotation and scale direction should continue with the model. NOT the build plate. 
[image]

BrittanGolden, 2019-12-31:
Oh, when you rotate any object and try to scale from it actual axis it skews the part in a diagonal angle instead of its original plane of axis. You should be able to rotate any object at any time and scale it without it turning into a parallelogram Trapezoid monstrosity haha.

Ghostkeeper, 2019-12-31:
> Oh, when you rotate any object and try to scale from it actual axis it skews the part in a diagonal angle instead of its original plane of axis. You should be able to rotate any object at any time and scale it without it turning into a parallelogram Trapezoid monstrosity haha.

Ah, this is a duplicate of https://github.com/Ultimaker/Cura/issues/4136.

Ghostkeeper, 2019-12-31:
For future reference, it helps a lot to
* Put every feature in a separate issue.
* Fill in the templates we provide. It is designed to force you to explain your feature in a way that we can understand what you mean and do something with it.

closed 2019-12-31 by Ghostkeeper

BrittanGolden, 2019-12-31:
I’m glad it’s a previously spoken issue. However your ref is from a
something posted about 3.4.1
1. You clearly understood why I meant not sure why you need to quote format
jargon like a machine.
2. How long has it been since reported and why for the last year it’s not
been fixed.
3. Do something other then pretend like I’m a novice for not already
searching a thread from countless patches ago assuming you would have fixed
it had it been identified so long ago.

Ghostkeeper, 2019-12-31:
I understood what you meant with the last one. Sorry, I picked that one out.

I don't know what you meant with numbering the build plate. There's only one build plate in Cura, so I can call the build plate number 42 but that won't help. Maybe you meant there to be some sort of number in the screen that changes every time you switch printers so you can identify the printer. Maybe you meant us to mark the 3D models of the build plates (for printers that have them). Maybe you meant us to show a number that increments every time you clear the build volume as to indicate a print. Maybe you meant us to actually print either of these numbers instead of displaying them in the interface.

I understand that duplicating the support blockers (void boxes?) may turn them into normal cubes, though I think that's a bug rather than a feature request since duplicating an object should also duplicate all of its per-object settings. I don't know what you meant with "to take advantage of your angle degree scale to change the support dynamic" or what this all has to do with the process of rescaling your model.

I don't understand what you meant with the last and very important thing either. If you scale an object there should be no change in its rotation. The scale has no direction unless you scale non-uniformly. The scale doesn't "continue" with anything as far as I know. I don't understand what you meant with that either.

We have a bug template and a new feature template for this reason, and it says that we'll close the issue if you don't fill it in. I'm normally quite lenient with that but I really had to use that disclaimer now, sorry.

BrittanGolden, 2020-01-01:
Okay, so when you rotate and object. The X Y and Z axis stays true to the
original loaded model. So if I spin it on it’s center axis the opposite
axis should follow with the model. As it is right now if I rotate a model
90’degrees it now is off axis and I can change any other axis with out it
going at an angle because the model moves but the rotation axis doesn’t.
For instance if I’m trying to make that Sith acolyte model stay in the
build plate I would put it at a 45 and then rotate it up or down to achieve
the desired angle of approach for less overhangs and fitting reasons. I
instead have to rotate the model off the build play first making it “non
printable” then rotate it to the 45 but disabling snap rotation I have to
go back and forth to level it over and over again because the model loses
the planes once it’s rotated.

With that being said, the scale feature does the same thing.. if you scale
the Z or X or Y then rotate it.. the program still scales at the original
angle.. making weird disproportionate angles that are just silly lol.

Also, being this is likely the only actually feature request.. on the build
plate it’s just empty squares. I make per person size items for people
weekly. If I could make a arm plate or a helmet/Mask having tiny numbers
showing the length and with from X to Y like a graph would make my life
SOOOO much easier.
 As for the programming you can just ad the feature and make the scale of
the numbers infinant so no matter what build volume is introduced to it
such as a CR-10S or an Ender 3 it would have the numbers of the X and Y by
the color they are indicated by from the home area and just make it a
toggle feature. So then I can literally just look right at it and know
exactly by the number how big it is. Instead of NS ing to painfully count
little white squares from one side of the model to the other.

Apologies for my rudeness and short nature in the last email. REALLY NICE
to have someone who actually took the time to get back with me.

BrittanGolden, 2020-01-01:
Side note. I’m aware the scale of the model is on the left side. The issue
is, not all models can be printed as One whole piece. Perfect example, I
was commissioned to make a 24 inch long Imperial Star destroyer. The only
way is to sink it 2/3rds of the way into my build platform and multiply the
model then invert it and print the other half in reverse to then glue
together later. The problem is the Z goes negative for both and no mater
what math I end up doing it’s never the exact length lost vs gained. That
issue comes up on the X and Y when you have to rotate the object a specific
way then try to scale it to your customers need. It’s always a guessing
game. I’m pretty good at it but if you have a solution I’m not seeing or
how to read the negative axis correctly..or maybe there’s a feature that
cuts it off at the build plate I don’t know about.

Ghostkeeper, 2020-01-01:
I was also a bit rude last time and I'm sorry for that. Doesn't take away that we really need to keep issues separated from each other to properly track which of them are still an issue and this thread is a perfect example of that problem.

> Okay, so when you rotate and object. The X Y and Z axis stays true to the
original loaded model. So if I spin it on it’s center axis the opposite
axis should follow with the model. As it is right now if I rotate a model
90’degrees it now is off axis and I can change any other axis with out it
going at an angle because the model moves but the rotation axis doesn’t.
For instance if I’m trying to make that Sith acolyte model stay in the
build plate I would put it at a 45 and then rotate it up or down to achieve
the desired angle of approach for less overhangs and fitting reasons. I
instead have to rotate the model off the build play first making it “non
printable” then rotate it to the 45 but disabling snap rotation I have to
go back and forth to level it over and over again because the model loses
the planes once it’s rotated.

Maybe I'll need some clarification on what the "center axis" is, or its opposite axis. But if I understood you correctly, the issue you're reporting there is that while the model is not fully inside the build volume, it won't drop down to the build plate automatically. This is actually intended because when you load a very big model in where the units are not in millimetres, or if you scale it to very big, the model would often be invisible if the bottom side is not completely solid, since everything would be above the camera's view. The middle of the object is almost always solid, so disabling the drop to build plate when the model is outside of the build volume helped people to not lose their 3D models by accidentally scaling it too much.

> With that being said, the scale feature does the same thing.. if you scale
the Z or X or Y then rotate it.. the program still scales at the original
angle.. making weird disproportionate angles that are just silly lol.

I think you're talking about the bug that this issue was marked a duplicate of, #4136 .

> Also, being this is likely the only actually feature request.. on the build
plate it’s just empty squares. I make per person size items for people
weekly. If I could make a arm plate or a helmet/Mask having tiny numbers
showing the length and with from X to Y like a graph would make my life
SOOOO much easier.

Good idea! Maybe we should open up a feature request about this.

> As for the programming you can just ad the feature

Should've thought of that! :stuck_out_tongue_winking_eye: 

But yeah we wouldn't normally give that sort of thing any limits. Our programming language handles big numbers well so there's no reason to not make these numbers count up to infinity if the printer is that big. Normally though most printers don't get much bigger than 400mm and that's a very easy number to count to for a computer.

> Side note. I’m aware the scale of the model is on the left side. The issue
is, not all models can be printed as One whole piece. Perfect example, I
was commissioned to make a 24 inch long Imperial Star destroyer. The only
way is to sink it 2/3rds of the way into my build platform and multiply the
model then invert it and print the other half in reverse to then glue
together later. The problem is the Z goes negative for both and no mater
what math I end up doing it’s never the exact length lost vs gained. That
issue comes up on the X and Y when you have to rotate the object a specific
way then try to scale it to your customers need. It’s always a guessing
game. I’m pretty good at it but if you have a solution I’m not seeing or
how to read the negative axis correctly..or maybe there’s a feature that
cuts it off at the build plate I don’t know about.

The dimensions in the bottom left should be accurate to a tenth of a millimetre, which is enough to cut your model down to the exact layer. We're aware that this is a bit unwieldy. Some research has been made to allow cutting model in pieces easier, but we haven't had success yet (nor the opportunity for much research. We've always said though that Cura is not a modelling tool and you should really be doing this in modelling software like Blender.

BrittanGolden, 2020-01-02:
Well, the think I’m talking about is simple, though it may be by design I
don’t know. The not dropping to build plate is normal and I totally
understand that. I guess by center I mean this precisely. When the object
loads in the build platform and I’m modifying the size or angle is the
issue.
The original picture I sent is the example. The X and The Y are no longer
with the model once rotated.
Put your hand on your desk, the planes of movement are indicated by the
colored circles representing them (imagined) turn your hand on the desk so
it’s facing at a 45 degree angle. Where should the adjustable circles be
after the manipulation? With the build platforms XYZ or the model? As it is
it literally does not move once rotated. So, if I spin an object on its
center just like you did your hand the circles of manipulation should move
with THE MODEL lol. Not stay as the build plates.
 What happens is I go to level the item on the play after rotation to
lessen support generation.  Go back to your hand and instead of lifting
from wrist and palm to tip of fingers like it should it instead lifts from
thumb to pinky totally ignoring the 45 degree angle and just lifting from
the built plates orientation instead.
 This issue only Occurs when the rotations is anything but 90 or 180 and
you can just use the opposite planes of movement as if they were there
counterparts. However the other 358 degrees of circular motion are now not
able to be leveled using the rotation as it will now lift from odd angles
and have to be reset and leveled before being rotated.