Ultimaker/Cura #6891: [4.4.1] Build Plate Adhesion Extruder doesnt't work with Skirt

siteswapjuggler, 2019-12-31:
**Application version**
4.4.1

**Platform**
Windows 10 or Mac OS

**Printer**
Tried with CR-X, Ultimaker S5, custom 4 extruder custom FFM

**Reproduction steps**
1. Get a multimaterial project
2. Choose a skirt option and a specific extruder to do the skirt

**Actual results**
The skirt is a combination of all extruders.

**Expected results**
The skirt should be done by only one extruder.

Ghostkeeper, 2019-12-31:
This is intended behaviour. In order to prime each nozzle correctly, it's going to draw a line around the skirt/brim with each extruder separately.

If this is not desired, reduce the Skirt/Brim Minimal Length to 0.

closed 2019-12-31 by Ghostkeeper

siteswapjuggler, 2019-12-31:
Seems a perfect way to do it, thanks for the answer, still a little bit confusing that each extruder prompt the same material instead of prompting there own

1013, 2020-03-05:
> This is intended behaviour. In order to prime each nozzle correctly, it's going to draw a line around the skirt/brim with each extruder separately.
> 
> If this is not desired, reduce the Skirt/Brim Minimal Length to 0 for that extruder.

This is perfect....I have been looking for this work around for weeks. any chance it can be added to the pop up instruction box for "build plate extruder" as that is what is expected.