Ultimaker/Cura #6893: [4.4.1] Wifi disconnect/reconnect randomly when cura is started

ERICRAC, 2019-12-31:


ERICRAC, 2019-12-31:
**Application version**
4.4.1

**Platform**
Windows 10

**Printer**
Creality ender 3

**Reproduction steps**
1. Just start Cura software => Issue
2.  Just close Cura software => Wifi is stable

**Project file**
Nothing

ERICRAC, 2019-12-31:
I just upgrade Cura 3.x to 4.4 on windows 10 and the same network (wifi drops) is appearing. I already had network issue with 3.x but not so much and didn't make relation with Cura.
That time with 4.4 started wifi loose connection very soon and wifi is back to life when Cura is closed. It looks that Cura is changing computer network configuration.
Could you help please?

**Application version**
4.4.1

**Platform**
Windows 10

**Printer**
Creality ender 3

**Reproduction steps**
1. Just start Cura software => Issue
2.  Just close Cura software => Wifi is stable

**Project file**
Nothing
>

fieldOfView, 2019-12-31:
Can you try disabling the "Ultimaker Network Connection" plugin on the "installed" tab of the "Marketplace" and restarting Cura?

Does this happen when Cura before Cura starts slicing? Do you have a VPN or are you using antivirus software that might set a default gateway for your network?

Do you use the OctoPrint Connection plugin?

Ghostkeeper, 2020-01-01:
See also https://github.com/Ultimaker/Cura/issues/4588.

I haven't been able to reproduce this problem so far. I use Cura every day and my work laptop is connected via wifi. Though there I am using Ubuntu, so maybe it doesn't give a problem on that OS.

ERICRAC, 2020-01-01:
**"Ultimaker Network Connection"** doesn't exist in my list. I've **Ultimaker Machine Actions** and diseable it for test without result. Same test and result by unchecking **UM3 Network prinitng** and **update checker** and **Firmware updater** and **Firmware Update Checker**
**Does this happen when Cura before Cura starts slicing?** It happen just after launching Cura without any action. Wifi disconnect when Chrome make a request (winodws refresh and google search request...
**Do you have a VPN or are you using antivirus software that might set a default gateway for your network?** No nothing special. At network conf level, Virtual box and VMWare are installed but not launch at all!
**Do you use the OctoPrint Connection plugin?** No
I've also stop any possible running software on windows without any resolution of the issue

Ghostkeeper, 2020-01-02:
> I've **Ultimaker Machine Actions** and diseable it for test without result.

This plug-in adds some extra machine actions like a manual bed levelling wizard. It doesn't have anything to do with network connection.

Plug-ins that do anything with the network are:
* UM3 Network Printing. When starting Cura it'll look for printers on the network, and various other things when connected to an Ultimaker printer.
* Cura Backups, but only when you open its dialogue and log in with it.
* Firmware Update Checker, upon starting Cura if the "check for updates" preference is enabled.
* Slice Info, every time you save the g-code or send it via wifi, sending anonymous statistics if the preference is enabled.
* Update Checker, upon starting Cura if the "check for updates" preference is enabled.
* Various plug-ins in the Marketplace such as Octoprint and 3DPrinterOS.

However Cura itself has some functionality that isn't in a plug-in that'll also make a connection. In particular, the Marketplace checks for updates to plug-ins and material packages if the "check for updates" preference is enabled, and a crash report is sent if you click on that button when Cura has a crash.

ERICRAC, 2020-01-06:
Ok but it doesn't solve my issue.
I've hundred of software installed on my computer with database, Upnp DNLA server, Java...
I never seen this behaviour before.

Ghostkeeper, 2020-01-13:
Probably some of that software also uses Qt to make the HTTP requests. Maybe the problem is in PyQt5? PyQt is quite rare, out in the wild, for a big application. But I've ran Cura thousands of times and have never seen this behaviour before either.

> Wifi disconnect when Chrome make a request (winodws refresh and google search request...

Could the problem lie in Chrome then?

ERICRAC, 2020-01-17:
My network issue is gone, I don't know why????? I'd a windows update in the meanwhile !!

New : I have to start it as admin to see UI. Otherwise UI is not visible but process running.

mahtDFR, 2020-01-20:
Good to hear it's working again. Please open a separate issue to discuss your new issue, quoting this one for reference if necessary.

closed 2020-01-20 by mahtDFR

rifqisetiawaan, 2021-02-26:
Hi, currently use cura and i found this slicer very powerful. but i encountered a problem which is my laptop's wifi network just suddenly disconnected immediatly when i used cura

Application version
4.8.0

Platform
Windows 10

Printer
Creality ender 3

Reproduction steps

Just start Cura software => Issue
Just close Cura software => Wifi is stable
Project file
Nothing