Ultimaker/Cura #6894: fusion 360 cannot open cura 4

jbvia, 2019-12-31:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-01-01:
Fusion 360 can't open Cura.exe since it's an executable file, not a model file.

I think you're referring to the "print" functionality of Fusion as demonstrated [here](https://www.youtube.com/watch?v=SgoMtf9fEAw). Though of course I can only guess since there are no reproduction steps. It would help if you give a bit more information.

I don't have Fusion installed so I can't try. However I tested the command that Fusion supposedly calls: `Cura.exe --single-instance model.stl`. That seems to work fine still.

rburema, 2020-01-02:
This is a duplicate. (See #6697) The problem is on Fusions' side, not ours. Please report it to them if this persists.

From the linked original:

> By default, this key does not have the --single-instance flag.   Attempting to add the flag to the registry does not help.   If it is after the "%1" value, Fusion360 will ignore it and call Cura without the flag.   If it is before the "%1" flag, Fusion360 will fail to launch Cura.
>
> In short, Fusion360 needs to change its Cura integration code to
>
> 1.) Obtain the location of Cura from the registry key used by current versions (Cura.model.STL instead of Cura STL model file)
>
> 2.) Append " --single-instance" to the value retrieved from the registry when invoking Cura.
>
> That should be an easy fix for the Autodesk folks to make.

closed 2020-01-02 by rburema