Ultimaker/Cura #6895: [4.4.1] some holes are blocked in printing and in previewing - in prepare they open

guy-maker, 2020-01-01:
**Application version**
4.4.1

**Platform**
windows OS 64 bit 

**Printer**
Creality Ender Pro 10 

**Reproduction steps**
1.open STL file and rotate it so it lies on the floor 
2. slice 
3. preview 

you can see that after preview some holes are blocked by filament - and also when printing - these would be blocked

[image]

[image]

smartavionics, 2020-01-01:
This is very likely to be a problem with the model. Please save the project file (File -> Save) and then zip (or rename to .zip) the .3mf and attach to this issue. Thanks.

guy-maker, 2020-01-01:
[back7.zip](https://github.com/Ultimaker/Cura/files/4014013/back7.zip)

attached the model

smartavionics, 2020-01-01:
Thanks for model, it has problems as you can see from the mesh tools plugin message...

[image]

guy-maker, 2020-01-01:
was issue with the model - don't know what issue exactly - I redraw it again and it is ok.

closed 2020-01-01 by guy-maker