Ultimaker/Cura #6896: Wrong Slicing on Cura 4.4.0/.1

Eagle71, 2020-01-01:
**Application version**
Cura 4.4.0 4.4.1

**Platform**
Win7 64Bit

**Printer**
Creality Ender 3

**Reproduction steps**
1. Lode Model
2. Rotate Vertical
3. Slice

**Screenshot(s)**
Cura4.4.x 
Blue Ground over Support not flat, makes a V.
Red Wrong Form
[image]

Correct in Cura 4.0.0
[image]

Wrong, building a wall, no Air goes out of the jet
[image]

**Actual results**
Wrong printing  ;-)

**Expected results**
Use older Cura

**Project file**
[HM_Volcano_Single_5015_Duct_Gen2e.zip](https://github.com/Ultimaker/Cura/files/4014093/HM_Volcano_Single_5015_Duct_Gen2e.zip)

**Log file**
[cura.zip](https://github.com/Ultimaker/Cura/files/4014125/cura.zip)

**Additional information**
Sorry, no more Info

smartavionics, 2020-01-01:
The model isn't perfect as you can see here...

[image]

However, when I slice it using my Cura build it doesn't appear to have any obvious problems...

[image]

Eagle71, 2020-01-01:
I added a printer: Ultimaker 2+
There all is fine.
Switched back to Creality Ender 3 (pro) and it is bad. So there musst be the fault!
Added a second Ender 3 
There is all fine.
I tested.......It is my faster Profile. Can you find the reason why this happens?

My Profile
[PETG-Faster-Std-FAULT.zip](https://github.com/Ultimaker/Cura/files/4014426/PETG-Faster-Std-FAULT.zip)

Ghostkeeper, 2020-01-01:
The model is indeed not watertight:
[image]
However that doesn't seem to be the problem. It slices fine.

The problem is that [Make Overhang Printable](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/experimental/conical_overhang_enabled.md) is enabled in your profile. This model is hollow and at the top side of the cavity there is a lot of overhang. That setting adjusts your model so that there is no more overhang. Probably not what you want, but expected behaviour. Solution: Uncheck that setting.

closed 2020-01-01 by Ghostkeeper

Eagle71, 2020-01-01:
Thanks for the help to solve the problem.
It's not a problem, it's a feature

The model is not the problem! I printed it with Cura 4.0.0. 
It look not easy, but it spend some  support blocker, that it keept hollow.
And the printer bridges the overhang.
It's only the PETG little bad, so the beginning layers over the support was bad.

[image]