Ultimaker/Cura #6898: [4.4.1] ignoring the nozzle and bed temperatures for custom materials

jamieturnough, 2020-01-02:
**Application version**
[4.4.1]

**Platform**
Windows 10 printing via USB connection

**Printer**
Anycubic 4 Pro (using 4max settings as printer)

**Reproduction steps**
I've duplicated a generic PETG and customised the settings to account for CF-PETG for printing draft (0.3mm). 
I adjusted the nozzle temp up to 250c and bed to 80c. 

When sending to the printer, it is adding 35c to the temperate and ignoring the bed temp setting. i.e. tries to print at 285c with bed at 70c.

Tried using a default Generic PETG, which should print at 220 with a bed of 80x, but it still prints at 255 + 70 bed. 

Note. that my printer will freeze when it tries to heat a nozzle over 265c (it just freezes) needing a hard reset. I assume this is a thermal cutoff. this might be causing some other issues mentioned in the forum.

**Screenshot(s)**
Note required.

**Actual results**
sets nozzle temp to 255 and bed at 70c.

**Expected results**
should use default temps of 220c + 80c

**Project file**
not required. error in your config files (see below)

**Log file**
not relevant, error in your default config (see below)

**Additional information**
I traced the problem to be in your config settings for Anycubic Max as its adding 35c and defaulting the bed to 70c.

**anycubic_4max_petg_draft.inst.cfg**
[general]
version = 4
name = Draft
definition = anycubic_4max

[metadata]
setting_version = 10
type = quality
quality_type = draft
weight = -2
material = generic_petg

[values]
material_print_temperature = =default_material_print_temperature + 35
material_bed_temperature = 70
cool_fan_enabled = False

speed_print = 30

Note sure why you would want to do this so just reporting it as an error?

rburema, 2020-01-02:
> I traced the problem to be in your config settings for Anycubic Max as its adding +35c and defaulting the bed to 70c. file below

When it comes to 3rd party printers, we have to rely on the community. If anything is wrong with the profile there, please make a pull request. (Or if it's just a few values, tell us what you think they should be.)

Ghostkeeper, 2020-01-02:
I made some adjustments to remove the 70 degree bed temperature override. For almost all profiles it was just the same as the temperature in the material profile. For one material it wasn't, so I moved it as an exception in that material so that it still uses 70 degrees for Anycubic 4Max printers then. This should allow you to change the bed temperature then.

The printing temperature seems to be done explicitly. I'm leaving that in.

We can't change the actual profiles (aside from maintenance) since we can't test with the printers themselves. So we won't change this for you. If you've made changes to the built-in profiles and tested them with the printer, drop us a pull request and we'll take a look at it!

closed 2020-01-02 by Ghostkeeper

jamieturnough, 2020-01-03:
Understood thanks. 

 �

I am using the Anycubic 4Pro, not the 4 Max (which isn’t listed) so I cant test with the exact print. As such, I’ll just adjust my local config file to remove the +35’c.

 �

One last question. Is there a reason why the nozzle doesn’t start heating up until the bed is at temp? Not a major problem but add a couple of extra minutes to every print.

 �

Thanks for the response.

 �

Jamie

 �

 �

Ghostkeeper, 2020-01-03:
Some printers don't take heating both at the same time, frying the power supply. Heating one before the other is a safe default. The bed should be done first then, so that the nozzle is not hot for an extended amount of time without extrusion, which would cause degradation, clogged nozzles and oozing.

This can be changed by adding the temperature commands to the start g-code. Cura will not put its own temperature commands in if the replacement keys are used like `{material_print_temperature}`.