Ultimaker/Cura #6899: [4.4.1]Duplicating material presets copies wrong data

Marlor, 2020-01-02:
When you duplicate a generic material, the copy will get wrong data.
This happens with some presets like PLA, PETG and ABS but not all of them like PC and PVA

**Application version**
4.4.1

**Platform**
Windows 10

**Printer**
Creality CR10s

**Reproduction steps**
1. Klick on Duplicate on a Preset like PLA (Build Plate Temp is 60°C)
[image]

2. You will get a copy were Build Plate Temp is 50°C, retraction is different too
[image]

rburema, 2020-01-03:
Seems to happen when a printer of that type is active.

rburema, 2020-01-03:
We've made a ticket for it in our internal system: CURA-7087

evtrados, 2020-01-15:
The issue has been resolved and the normal behavior of the material duplication will be working in the next Cura release.

closed 2020-01-15 by evtrados