Ultimaker/Cura #6900: [4.4.1] Only One Layer Being Sliced

blkexp98, 2020-01-02:
**Application version**
4.4.1

**Platform**
Windows 10 Pro
Ryzen 3800X CPU
32 gigs Ram
Nvidia (EVGA) RTX 2070 Super

**Printer**
Primary: Creality Ender 3 (Also did troubleshooting with a Ultimaker 3 profile)

**Reproduction steps**
1. Tried reinstalling CURA and using the default settings for an Ender 3 including the slicing settings.  I downloaded a new STL file from the Ultimaker site since I knew it wouldn't have print issues and simulated the highstand portion of the wind turbine.  All it would simulate was the top slice.  The funny thing is it will show the object in all yellow when I click slice but when I go to preview it only simulates the top slice.  Also if you look at the time to print it looks about right for the whole part.  However if you click on the ! the time looks about right for just that one layer.    
2. I repeated the above steps without reinstalling but used the ultimaker 3 printer as the profile in case it was an issue with the Ender 3 profile.  The results were the same.   

**Screenshot(s)**
Ultimaker 3 profile (Everything unmodified)
[image]

Ender 3 profile (Everything Unmodified)  Note the print time and the expanded print times.
[image]

**Actual results**
Only the top layer appears to be simulated.  

**Expected results**
For the simulation to show each layer of the print instead of just the top layer.

**Project file**
[HighStand.zip](https://github.com/Ultimaker/Cura/files/4016070/HighStand.zip)

**Log file**
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/4016109/DxDiag.txt)
[cura.log](https://github.com/Ultimaker/Cura/files/4016128/cura.log)
[stderr.log](https://github.com/Ultimaker/Cura/files/4016131/stderr.log)

**Additional information**
I have tried multiple STL files with the same result so I dont think its tied to the STL itself.   I use octoprint so it isnt an SD card issue either as for the most part im not eve using that step and this is all taking place before that anyway.

smartavionics, 2020-01-02:
And what happens when you move the two layer sliders on the right?

blkexp98, 2020-01-02:
Humm...  It will simulate any layer I stop on there but only the top one.  So say I click on the bottom circle where it will show layer 1 on that slider it will still only simulate the uppermost layer I have selected on that slider.  So if I set the range to say 1 to 461 then it will only simulate layer 461.  If i move the slider up from 1 then the layers start to get greyed out and it doesnt show them as layers.  Only a transparent model.

rburema, 2020-01-03:
The lower horizontal bar is the simulation of the currently selected layer.
The vertical bar on the right is the layers.
If you press the play-button on the left of the horizontal bar, it should also move up through the layers (resetting the horizontal bar to the beginning, and adding 1 to the vertical bar).

closed 2024-12-07 by github-actions[bot]